dronekit/dronekit-sitl #3: plane should also to be considered

mrpollo, 2015-06-10:
those fixed wing dudes love them dronekits

hamishwillee, 2015-07-07:
"Must" also be considered

tcr3dr, 2015-07-07:
@mrpollo @hamishwillee I've added support for plane binaries that are added to S3. This works on master:

```
$ dronekit-sitl plane-3.3.0 --help
os: osx, apm: plane, release: 3.3.0
SITL already Downloaded.
Execute: ['./plane-3.3.0/ArduPlane.elf', '--help']
Options:
Improves README.
	--home HOME        set home location (lat,lng,alt,yaw)
	--model MODEL      set simulation model
	--wipe             wipe eeprom and dataflash
	--rate RATE        set SITL framerate
	--console          use console instead of TCP ports
	--instance N       set instance of SITL (adds 10*instance to all port numbers)
	--speedup SPEEDUP  set simulation speedup
```

Now, I've only compiled binaries for ArduPlane 3.3.0. Should I also compile 3.2.x at all?

hamishwillee, 2015-07-08:
@tcr3dr It would be good if we could have plane versions that match Copter releases.

Can we please have Rover too?

closed 2015-09-11 by hamishwillee