dronekit/dronekit-sitl #10: Linux connection advice

hamishwillee, 2015-07-07:
I have been able to test our installation instructions "a bit" on linux. I opened a recent Linux ubuntu build in a VM and installed the dronekit-sitl runner exactly as in the instructions (except using "sudo" prefix).

Once running, the dronekit-sitl-runner gets to a point as shown:
```
Started model quad at -35.363261,149.165230,584,353 at speed 1.0
Starting sketch 'ArduCopter'
Starting SITL input
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
```

I then used ``ifconfig`` to get the IP address of the VM and was able to connect over TCP from Mission Planner running on the host windows computer (to port 5760).

However I can't seem to connect to the instance using UDP from the host or the VM. I've tried 
mavproxy.py --master=127.0.0.1:14550
and 
mavproxy.py --master=192.168.65.133:14550  (ip address of the VM)
and
mavproxy.py --master=192.168.65.133:5760 (the TCPIP port)

I also changed the output of the SITL to be the IP address of the host.

Does this work/how do I get this working? Basically I need to get the linux->linux working so that the dronekit instructions make sense.

tcr3dr, 2015-07-07:
@hamishwillee This would be the biggest "gotcha" when using dronekit-sitl directly, in that it does not replicate the full functionality of [sim_vehicle.sh](https://github.com/diydrones/ardupilot/blob/master/Tools/autotest/sim_vehicle.sh).

SITL only exposes that TCP connection. sim_vehicle spawns SITL, but then also an instance of mavproxy that forwards TCP -> UDP, similar to:

```
mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551 
```

So to create a UDP connection exposed on Linux, or exposed through a VM, you'd need to explicitly run a separate mavproxy instance forwarding the tcp:localhost:5760 connection.

For Dronekit users, perhaps we want to incorporate into dronekit-sitl-runner some functionality provided by `sim_vehicle.sh`? E.g. creating UDP connections and some hardcoded locations (like "3drberkeley"). The rest of `sim_vehicle.sh` feels redundant though.

hamishwillee, 2015-07-08:
@tcr3dr Thanks very much. The problem for me here was the readme.md text - I interpreted that all as one line (ie passing mavproxy as part of the command line to dronkit-runner). Instead this is two separate lines of text that need to be run in separate terminals/prompts. 

Your explanation of what the second line does (and why) also very useful.

It would be good if the readme.md made this clear, or even better if the dronekit runner automatically spawned the mavproxy instance?

> The rest of sim_vehicle.sh seems redundant, and is obviously bash-specific.

Yes.

hamishwillee, 2015-07-09:
Closing this as the question is answered. I've created #16 to address my last points.

closed 2015-07-09 by hamishwillee