dronekit/dronekit-sitl #9: Windows: 'dronekit-sitl' is not recognized as an internal or external command

hamishwillee, 2015-07-07:
I followed the instructions on windows in a command prompt (ActiveState Python 2.7.8). The runner appeared to install fine but I got the above error when try to run SITL.

TShapinsky, 2015-07-07:
This will happen if your python "Scripts" folder isn't included in your windows PATH variable. The only fix I know is to add it manually.

hamishwillee, 2015-07-08:
That isn't the problem - Scripts is in my path. The "first" problem is that windows interprets how to run a file based on its extension. This has no file extension so windows does not need to execute it. If I change the extension to ".py" it can be executed ...

But then I get the following error:

```
C:\Users\hamis_000>dronekit-sitl copter-3.4-dev
Traceback (most recent call last):
  File "C:\Python27\Scripts\dronekit-sitl.py", line 7, in <module>
    from dronekit_sitl_runner import *
ImportError: No module named dronekit_sitl_runner
```

@tcr3dr Any ideas?

tcr3dr, 2015-07-08:
@hamishwillee I can recreate the issue. I'm following along with https://matthew-brett.github.io/pydagogue/installing_scripts.html and will be working on the `tcr-windows` branch to fix it.

tcr3dr, 2015-07-08:
@hamishwillee Wrestled with CI but got a fix merged. Can you test out master (with the python scripts directory in your path) and see if `dronekit-sitl` (no .py extension) now works?

hamishwillee, 2015-07-08:
@tcr3dr I wasn't sure how to test this "properly" so I just re-ran the instructions in readme. This appears to have uninstalled the old version and re-installed correctly.
```
Successfully installed Dronekit-SITL-Runner
Cleaning up...
```

Windows can now find dronekit-sitl (without extension). SITL still doesn't start though. Output from command line below:

```
C:\Users\hamis_000\Documents\vagranttesting\dronekit-sitl-runner>dronekit-sitl copter-3.4-dev
os: linux, apm: copter, release: 3.4-dev
Downloading SITL from http://d3jdmgrrydviou.cloudfront.net/copter/sitl-linux-v3.4-dev.tar.gz
Extracted.
Execute: ['./copter-3.4-dev/ArduCopter.elf']
Traceback (most recent call last):
  File "C:\Python27\Scripts\dronekit-sitl-script.py", line 9, in <module>
    load_entry_point('Dronekit-SITL-Runner==1.3.0', 'console_scripts', 'dronekit-sitl')()
  File "C:\Python27\lib\site-packages\dronekit_sitl_runner\__init__.py", line 113, in main
    launch(system, version, args)
  File "C:\Python27\lib\site-packages\dronekit_sitl_runner\__init__.py", line 60, in launch
    p = Popen(args, cwd=sitl_target)
  File "C:\Python27\lib\subprocess.py", line 710, in __init__
    errread, errwrite)
  File "C:\Python27\lib\subprocess.py", line 958, in _execute_child
    startupinfo)
WindowsError: [Error 2] The system cannot find the file specified
```

hamishwillee, 2015-07-12:
@TShapinsky @tcr3dr Is it time for me to re-test this on Windows (I saw #15 was closed). Should I close this and reopen a new issue for whatever issue is detected?

TShapinsky, 2015-07-12:
@hamishwillee if it gives you an error you should probably open up another issue. Make sure before you retest that you have deleted the ~/.dronekit/sitl folder since you probably have the linux binaries installed.

hamishwillee, 2015-07-13:
@TShapinsky Installing on windows and then running the following works, so I will close this:

```
dronekit-sitl copter-3.4-dev -I0 -S --model quad --home=-35.363261,149.165230,584,353
```

closed 2015-07-13 by hamishwillee

tcr3dr, 2015-07-16:
@hamishwillee Excellent!

hamishwillee, 2015-07-16:
@TShapinsky @tcr3dr Yes it is :-) I love this new tool. I think it will be ready to replace the vagrant option in docs when we've fixed #16 and #24.