dronekit/dronekit-sitl #12: What are the SITL options that can be passed to the dronekit runner

hamishwillee, 2015-07-08:
If you run ``dronekit-sitl`` or ``dronekit-sitl --help`` you get the following options:
```
ubuntu@ubuntu:~$ dronekit-sitl 
Please specify one of:
  dronekit-sitl --list
  dronekit-sitl --reset
  dronekit-sitl <copter(-version)>
  dronekit-sitl <plane(-version)>
```

However I can see from your example I can actually specify more options (model, home, IO, s):
```
dronekit-sitl copter-3.4-dev -I0 -S --model quad --home=-35.363261,149.165230,584,353
```
Is there a list of all options we can pass anywhere?

TShapinsky, 2015-07-08:
Everything after the copter or plane version is passed to ArduPilot. You can get a list of parameters by running `./Ardu<Copter/Plane> -h`

hamishwillee, 2015-07-09:
@TShapinsky Thanks. The readme and help should say that :-)

TShapinsky, 2015-07-09:
@hamishwillee because dronekit-sitl passes through all the arguments to Ardu<Copter/Plane> if you ran `dronekit-sitl copter-3.4-dev -h` it should give you a list of parameters because the available parameters change between different versions of Ardu<Copter/Plane>

hamishwillee, 2015-07-09:
Thanks very much. Closing this (have updated #17 appropriately)

PR #17 (2015-07-13): Add parameters and expand example section of README.md
This partially fixes #12 (it would still be good if the parameters could be fetched and displayed using `` dronekit-sitl --help``.

This fixes the example, which was previously ambiguous (not clear what the mavproxy command was doing, or that it needed to be run in a separate terminal)
closed 2015-07-09 by hamishwillee