dronekit/dronekit-sitl #16: dronekit-sitl-runner should spawn mavproxy to set up UDP

hamishwillee, 2015-07-09:
The readme.md example text looks like either just one line of text or two lines that should be used in the same prompt after each other:
```
dronekit-sitl copter-3.4-dev -I0 -S --model quad --home=-35.363261,149.165230,584,353
mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551
```
In fact the first line starts SITL which only exposes its TCP connection and waits.
```
dronekit-sitl copter-3.4-dev -I0 -S --model quad --home=-35.363261,149.165230,584,353
```
The second line spawns an instance of mavproxy that connects SITL TCP and forwards packets to  UDP localhost - effectively doing the same as `sim_vehicle.sh`. This is optional as you could just connect to the TCP port, but is useful because it allows SITL to be used via UDP in a familiar way.

```
mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551
```

The requirement is that  dronekit-sitle-runner spawn mavproxy for you. In this case the second line should be removed from the readme.

If this is not possible, then the readme.md needs to explain that these commands are run in different terminals/command prompts.

hamishwillee, 2015-08-11:
@tcr3dr I think this is the most important issue stopping me documenting dronekit sitl runner. IMO not having this makes it quite unfamiliar to users, and a little harder to explain how to set up. 

* Can we have agreement on whether this can be done/get resource
* Is there anything else that makes this "harder" than using the current vagrant and native solutions (ie any other reasons not to push this as our "main solution"?

mrpollo, 2015-08-12:
@hamishwillee I think running `MAVProxy` from the `dronekit-sitl-runner` should not be a feature of this tool, we need to keep `dronekit-sitl-runner` slim and fast, the main motivation to make this tool was that `sim_vehicle.sh` felt overly complex was hard to understand and that you need dependencies to run it, with `dronekit-sitl-runner` we want 0 friction to run this.

That said `MAVProxy` is a great tool and we should probably still support it, I think the best approach is to create a `MAVProxy` module that launches `dronekit-sitl-runner`, this of course would be a separate project that would use both `dronekit-sitl-runner` and `MAVProxy`.

tcr3dr, 2015-08-12:
@mrpollo See my comments here though: https://github.com/dronekit/dronekit-sitl/issues/20#issuecomment-121078906 If UDP is the requirement (which may be right) then it's not very difficult to add as a feature to sitl-runner directly.

mrpollo, 2015-08-12:
I was refering to "launch mavproxy from within dronekit-sitl-runner`

hamishwillee, 2015-08-13:
@mrpollo Yes, the original suggestion pre-dates @tcr3dr idea that we just add the support for UDP directly. 

I am suggesting we automate creation of the UDP port because every other SITL setup does so. Arguably our current audience won't have that expectation and this should be fixed in documentation. 

Do you think it is worth being "consistent"? If not, I'll just go ahead and document it as it is now.

mrpollo, 2015-08-13:
@hamishwillee Totally agree, its up to @tcr3dr, I would vote for yes?

hamishwillee, 2015-08-14:
@tcr3dr - document what we have now, or add support for UDP redirect in the tool? Nag!

cfotache, 2015-11-12:
I've been trying to set up the SITL so I can experiment with Python programs before buying the hardware, but I think I'm missing something. I followed the instructions here, and dronekit-sitl hangs on "Waiting for connection...." (using copter-3.3 instead of 3.4-dev that you mentioned here), while mavproxy hangs one "MAV> Waiting for heartbeat from COM3".
Are there are any more complete instructions on the steps needed to be able to get to the point where you can test the Python code?

hamishwillee, 2015-11-12:
@cfotache If it is giving you and error looking at COM3 then you're probably not connected to SITL at all - since that is on an IP network, and COM3 is a serial port (you would use to connect to the device).

You don't say if you're using Dronekit version 1 or 2, but in either case your string to connect to dronekit sitl should look like ``tcp:IPADDRESSOFHOSTRUNNINGSITL:5760``.  If you are running a separate GCS it can connect on port 5763.

Note, DK-SITL is experimental.

YOu can use the vagrant sitl described in the dronekit docs too - http://python.dronekit.io/guide/sitl_setup.html . IN this case the ip address string connect to is ``127.0.0.1:14550``

Let me know how you go!

cfotache, 2015-11-13:
I don't get the COM3 error anymore, but it doesn't seem to be connected.
I checked and I have installed Dronekit 2.0.0rc9 and Dronekit_sitl 3.0.1, also MavProxy 1.4.37 (all installed with pip).
I ran dronekit-sitl and mavproxy with the 2 commands at the top of this thread, in different windows.

Then I tried this from the Python interpreter:
from dronekit import connect
vehicle = connect('127.0.0.1:14550', wait_ready=True)
And I get the message of Link timeout, no heartbeat.

hamishwillee, 2015-11-13:
@cfotache The address `127.0.0.1:14550` is the loopback address for using UDP. This would work with the vagrant SITL described in the dronekit docs, but the dronekit-sitl does not support port forwarding or UDP (yet). 

The solution is to use a tcp connection like this: ``tcp:youripaddress:5760`` instead. YOu will need to get your ipaddress, which you can get by running ``ipconfig`` on the command line.

You can see the needed port when you start SITL:
```
...
Starting sketch 'ArduCopter'
Serial port 0 on TCP port 5760
Starting SITL input
Waiting for connection ....
```

MAVProxy will need to connect to another address, but will have to use a similar string. The address will be the same but the port will change (and you'll see this after you've connected dronekit).