dronekit/dronekit-sitl #19: Auto-load parameters

squilter, 2015-07-09:
After getting a sitl session set up, I kept getting the error "rc not calibrated".  Loading params with `param load <ardupilot>\Tools\autotest\copter_params.parm` fixed the problem.  It would be nice to have these auto-loaded.  If that is not doable, an extra sentence in the readme to remind users to load params would be good.

hamishwillee, 2015-07-09:
+1 for this on first startup. You don't want it on every startup of course, because this would over-write any parameters you made and want to keep. Perhaps some sort of command line option that we can recommend in the example?

hamishwillee, 2015-09-22:
Also turning off the arming check

hamishwillee, 2016-02-08:
@tcr3dr @squilter I am closing this because when I load solo and copter SITL I'm no longer getting those warnings and the ARMING_CHECK is zero. While I can not see where these are being set/unset in EEPROM  or `default.parms` creation this testing shows it must be happening somewhere!

Please reopen if you disagree (and Tim, love confirmation that this is being done somewhere). I have added #73 to document the config, since it differs from "normal" SITL.