dronekit/dronekit-sitl #20: Should MAVProxy work "out of the box" on your python command line after installing this?

hamishwillee, 2015-07-13:
The example suggests you first run 
```
dronekit-sitl copter-3.4-dev -I0 -S --model quad --home=-35.363261,149.165230,584,353
```
Then
```
mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551
```

The mavproxy call only works if mavproxy is installed. Should the example explain how to get mavproxy?

tcr3dr, 2015-07-13:
@hamishwillee So my question for you is, in what situations is UDP needed instead of just TCP?

If a UDP connection is useful in any situation dronekit-sitl is, I believe that `dronekit-sitl` can simply create a TCP->UDP tunnel by default. This would be extremely simple to author in Python without a whole MAVProxy dependency, provided this is the only additional functionality to the SITL runner we want.

hamishwillee, 2015-07-14:
@tcr3dr The MAVProxy UDP functionality is useful because it makes it easy to set up multiple GCS connections on the network - for example you can set up MAVProxy and Mission Planner watching SITL at the same time (or Tower if you like, too).

Personally I have found this useful while testing, and it is familiar enough to users that Mission Planner and APM Mission Planner 2 both have UDP and port 14550 set as their default configuration. How much of a requirement this is for other developers I don't know. 

One reason I suggested MAVProxy approach is that I know we will have to install it anyway and we haven't got a really good way of doing that in a Python command prompt on Windows. Currently we can use the installer version which is very much "preconfigured" or we can use WinPython. I'd much rather have it all installed in my ActiveState Python as part of installing dronekit. In other words, trying to sort out a "bigger picture" problem - but maybe that is not worth the cost.

You would be far better positioned than me to make a good decision on whether implementing in the SITL runner is the right approach.