dronekit/dronekit-sitl #24: Should not require python -m prefix to launch on MAC OS X

hamishwillee, 2015-07-13:
From [this comment](https://github.com/diydrones/dronekit-python/issues/198#issuecomment-120999908):

> To run dronekit-sitl I end up invoking python -m dronekit_sitl_runner.__init__ which isn't the prettiest (I need to invoke mavproxy similarly through python -m MAVProxy.mavproxy, but at least it looks nicer). The -m flag won't let me name a function to call, so I had to add the following to __init__.py:
....
After that, python -m dronekit_sitl_runner.__init__ --list works as expected.

The bug is that you should not need to run ``python -m dronekit_sitl``. Using ``python -m`` should of course be possible.

tcauchois, 2015-07-14:
This is just a result of how macports installs python packages

hamishwillee, 2015-07-14:
@tcauchois I know nothing about Mac or macports. But unless every script run on Mac needs the "python -m" I would expect it not to be needed for basic use of the runner. Ideally the instructions in the readme should work for all platforms (possible exception need for "sudo" on linux).

tcr3dr, 2015-07-16:
I'll look into this, given paths issues are going to be support questions regardless.

tcr3dr, 2015-07-30:
#25 will fix this to be invokable by `python -m dronekit.sitl`, which is a decent concession to Python's confusing install paths.

closed 2015-07-30 by tcr3dr

hamishwillee, 2015-07-30:
@tcr3dr I don't understand - it sounds like you're saying that this will still require ``python -m`` to invoke on Mac - which differs from the other platforms. Correct? If so, we need to update the readme.txt to explain this (and the changes that #25 requires)

tcr3dr, 2015-07-31:
@hamishwillee The issue is specifically with macports, which is not the suggested way to run python on OS X (the suggested way being using the built-in Python, or worst case brew works too).

I think we should see if there is ever another data point on this. Using `python -m dronekit.sitl` is an okay workaround if top-level installs don't work, but there are too many problems with custom Python installs on OS X to detail them all. ;) For what it's worth, `dronekit-sitl` installs on my machine and Travis just fine.