dronekit/dronekit-sitl #29: Error when downloading latests

hamishwillee, 2015-09-11:
I just ran:
```
D:\github\testsitlrunner>  dronekit-sitl copter-3.2.1
os: win, apm: copter, release: 3.2.1
Downloading SITL from http://d3jdmgrrydviou.cloudfront.net/copter/sitl-win-v3.2.1.tar.gz
Extracted.
```
At end got of this got error
```
Execute: ['.\\ArduCopter.elf']
'.\ArduCopter.elf' is not recognized as an internal or external command,
operable program or batch file.
```

tcr3dr, 2015-09-24:
I believe this is fixed; this was caused by inappropriately packaged tarballs. They are all rebuilt now.

hamishwillee, 2015-09-24:
After downloading new version I get a new error:

```
C:\Users\hamis_000>dronekit-sitl copter-3.2.1
os: win, apm: copter, release: 3.2.1
SITL already Downloaded.
Execute: ['.\\ArduCopter.elf', '--home', '-35.363261,149.165230,584,353']
ArduCopter.elf: unknown option -- -
Options:
        -w          wipe eeprom and dataflash
```

tcr3dr, 2015-09-24:
Hm. @hamishwillee try redownloading again, see if this fixed.

hamishwillee, 2015-09-24:
@tcr3dr Nope, same problem

hamishwillee, 2015-12-30:
Closed - you're no longer supplying these binaries.

closed 2015-12-30 by hamishwillee