dronekit/dronekit-sitl #30: Document how to point to local build in readme

hamishwillee, 2015-09-11:
[This comment](https://github.com/dronekit/dronekit-sitl/issues/13#issuecomment-127346813) shows how to set up a local build. This needs to be put into Readme

I'd do it myself but couldn't get things to work - #29

tcr3dr, 2015-10-20:
Note that as of https://github.com/dronekit/dronekit-sitl/pull/50 the format is documented in the README and changed to "is this a local path (leading . or /)"

hamishwillee, 2015-10-20:
@tcr3dr #50 doesn't appear to update the README file so where are instructions as per your previous note?

Chambana, 2016-01-04:
Please document how to load custom firmware in the README file, as I still don't see it.  I believe the syntax is just as below, but I'm having issues on OSX El Capitan getting it load:
sudo dronekit-sitl ./ArduCopter-v2.px4

tcr3dr, 2016-01-04:
Added by https://github.com/dronekit/dronekit-sitl/pull/64

hamishwillee, 2016-01-04:
@tcr3dr The binary identified by @Chambana is, I believe, the wrong type - **.px4**. Shouldn't it be an **ArduXxxx.elf**?

#64 Fix does not make this clear - perhaps (?):
```
dronekit-sitl /path/to/local/binary/ArduXxxx.elf [parameters]
```