dronekit/dronekit-sitl #33: Floating point exception running copter-3.2.1

hamishwillee, 2015-09-22:
```C:\Users\hamis_000\Documents\vagranttesting\ramonsitl>dronekit-sitl copter-3.2.1
os: win, apm: copter, release: 3.2.1
Downloading SITL from http://d3jdmgrrydviou.cloudfront.net/copter/sitl-win-v3.2.1.tar.gz
Extracted.
Execute: ['.\\ArduCopter.elf']
ERROR: Floating point exception
Starting sketch 'ArduCopter'
Starting SITL input

C:\Users\hamis_000\Documents\vagranttesting\ramonsitl>
```

tcr3dr, 2015-09-24:
This is probably related to the lack of home parameter. Is this something you can confirm?

If you can't recreate the floating point exception easily with a home parameter (a list you can try [is located here](https://github.com/diydrones/ardupilot/blob/master/Tools/autotest/locations.txt)) then I would consider this fixed. It's still possible to get these exceptions after long periods of use but that's more related to SITL's internals.

hamishwillee, 2015-09-24:
After downloading new version I get a new error:

```
C:\Users\hamis_000>dronekit-sitl copter-3.2.1
os: win, apm: copter, release: 3.2.1
SITL already Downloaded.
Execute: ['.\\ArduCopter.elf', '--home', '-35.363261,149.165230,584,353']
ArduCopter.elf: unknown option -- -
Options:
        -w          wipe eeprom and dataflash
```

closed 2015-09-24 by hamishwillee