dronekit/dronekit-sitl #35: SITL should be easy to fly "out of the box"

hamishwillee, 2015-09-23:
SITL doesn't have default params by default so you can't pass arming checks on first load and the default params won't allow arming or flight. 

The system should make this easy for you

* automatically load the autotest params for the vehicle on first load (only) - these include disabling the armed_check param. 
* There there should be some command line argument to allow a new set of params to be loaded on startup OR a way to restart/reset back to the defaults (but this shouldn't be called automatically after the first startup).

tcr3dr, 2015-09-24:
I think the interaction for this should be, you always have stock params loaded unless you specify a `--eeprom` (`--save`?) option that stores the eeprom in a local location. On first creation, default params are loaded, but on subsequent loads it is not.

tcr3dr, 2015-09-24:
Right now, eeprom.bin is in a fixed location (current dir) and contains the parameter history, so the above interaction is doable with a little ~studio magic~.

hamishwillee, 2015-09-24:
Happy enough with loading default params on first load. What if I want to reload them later because I've changed a param at some point and now I want to reset?

I am assuming that correct params get loaded for each platform.

tcr3dr, 2015-09-24:
By default, you would have fresh params on each load.

If you specify `--eeprom`, an `eeprom.bin` file gets saved in your current directory (where you run sitl). The first time it's loaded you get fresh params. Subsequent runs preserve parameter state. You can reset the parameters simply by deleting `eeprom.bin`, which would be recreated if you have `--eeprom` specified still.

hamishwillee, 2015-09-24:
So if I delete eeprom.bin the next time I load I'll get the fresh default parameters from the .parms file? 

And if I run --eprom then changes I make to params will be saved. HOwever the next time I run sitl, if I choose not to specify --eprom I will still get the eprom version, but I won't save subsequent changes to parameters. Is that correct.

Current SITL stores your settings unless you explicitly reset or reload new ones. 

My concern with this is that (as I understand it) to model a different environment you load different parameters. Will we still be able to somehow load new parameters sets and keep them so that people aren't forced to just use the one good set (doesn't matter much for DK, but might restrict other uses)?

This does feel a bit unfamiliar too - but I think we can live with it as long as it is clearly documented and we're making it easy for people to model different vehicles if they need to.