dronekit/dronekit-sitl #36: SITL print the whole URL for connecting

hamishwillee, 2015-09-23:
SITL starts by exposing a TCP port. Ideally you should be able to connect this using a localhost address, but this failed on windows. That means that the instructions to get the "full URL" for connecting end up needing to be quite complicated (e.g. on windows, got to commandprompt, run ipconfig, get current address then do ....)

It would be better if

* Localhost connection to the TCP port worked
* Rather than just printing the TCP port to connect on, the tool printed the full IP address and port to connect to.

tcr3dr, 2015-09-24:
Can do. Example dump would be:

```
Waiting for connection ....
tcp:127.0.0.1:5760
udpin:127.0.0.1:6760
```

tcr3dr, 2015-09-24:
`udpin` is equivalent to `udp` or no prefix in MAVProxy; still better to specify it usually.

hamishwillee, 2015-09-24:
The loopback does not work for tcp connections (don't ask me why). I always have to explicitly put the network address in. So above looks great, but I think you'd need to get address from the interface.

By contrast, loopback does work for udp. I would be happy with a fix whereby the loopback works, because it would make the docs a lot cleaner.