dronekit/dronekit-sitl #41: Getting "help" via ArduXxxx.elf not working

hamishwillee, 2015-09-24:
The -h option isn't working properly on builds I've tried on Windows (should be giving me the help options from ArduCopter.elf') . Could somehow be due to #40 except that the .elf file is being launched.
```
C:\Users\hamis_000>dronekit-sitl copter-3.3-rc11 -h
os: win, apm: copter, release: 3.3-rc11
SITL already Downloaded.
Execute: ['.\\ArduCopter.elf', '-h', '--home', '-35.363261,149.165230,584,353']
Options:
```

```
C:\Users\hamis_000>dronekit-sitl solo-1.1.40 -h
os: win, apm: solo, release: 1.1.40
SITL already Downloaded.
Execute: ['.\\ArduCopter.elf', '-h', '--home', '-35.363261,149.165230,584,353']
Options:
```

```
D:\deleteme>dronekit-sitl solo-1.1.40 -h
os: win, apm: solo, release: 1.1.40
SITL already Downloaded.
Execute: ['.\\ArduCopter.elf', '-h', '--home', '-35.363261,149.165230,584,353']
Options:
        --home HOME        set home location (lat,lng,alt,yaw)
```

tcr3dr, 2015-10-09:
@hamishwillee, this should be fixed by https://github.com/dronekit/dronekit-sitl/commit/27063f0196fecf7a2eb3dcc07fb07b4a137f2eb2

```
$ dronekit-sitl copter-3.3-rc11 --help
os: osx, apm: copter, release: 3.3-rc11
SITL already Downloaded.
Execute: ['./ArduCopter.elf', '--help', '--home=-35.363261,149.165230,584,353', '--model=quad']
Options:
	--home HOME        set home location (lat,lng,alt,yaw)
	--model MODEL      set simulation model
	--wipe             wipe eeprom and dataflash
	--rate RATE        set SITL framerate
	--console          use console instead of TCP ports
	--instance N       set instance of SITL (adds 10*instance to all port numbers)
	--speedup SPEEDUP  set simulation speedup
	--gimbal           enable simulated MAVLink gimbal
	--autotest-dir DIR set directory for additional files
```

hamishwillee, 2015-10-14:
Yes, worked!

closed 2015-10-14 by hamishwillee