dronekit/dronekit-sitl #42: Can't get any builds to work on Windows using new pip build

hamishwillee, 2015-10-02:
I uninstalled dronekit-sitl, then re-installed using ``pip2 install dronekit-sitl -UI``. Then called ``dronekit-sitl --reset``. I tried all the builds a number of times. I rebooted the computer. Repeatable results below. 

The following three downloaded and started fine (ie binding the port). However only one heartbeat was received in all cases on Mission Planner - ie failed with timeout.
```
dronekit-sitl copter-3.3-rc11
dronekit-sitl copter-3.3-rc5
dronekit-sitl rover-2.50
```

This failed with an exception:

```
>dronekit-sitl copter-3.2.1
os: win, apm: copter, release: 3.2.1
Downloading SITL from http://d3jdmgrrydviou.cloudfront.net/copter/sitl-win-v3.2.1.tar.gz
Extracted.
Execute: ['.\\ArduCopter.elf']
Starting sketch 'ArduCopter'
ERROR: Floating point exception
Starting SITL input
```

This one seems to have buggered up paths:
```
C:\Apps\WinPython-64bit-2.7.10.2\python-2.7.10.amd64>dronekit-sitl plane-3.3.0
os: win, apm: plane, release: 3.3.0
Downloading SITL from http://d3jdmgrrydviou.cloudfront.net/plane/sitl-win-v3.3.0.tar.gz
Extracted.
Traceback (most recent call last):
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Apps\WinPython-64bit-2.7.10.2\python-2.7.10.amd64\Scripts\dronekit-sitl.exe\__main__.py", line 9, in <module>
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\site-packages\dronekit\sitl\__init__.py", line 306, in main
    sitl.launch(args, verbose=True, local=local)
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\site-packages\dronekit\sitl\__init__.py", line 148, in launch
    elf = open(os.path.join(wd, args[0])).read()
IOError: [Errno 2] No such file or directory: 'C:\\Apps\\WinPython-64bit-2.7.10.2\\settings\\.dronekit\\sitl\\plane-3.3.0\\.\\ArduPlane.elf'
```

This one failed to download:
```
>dronekit-sitl solo-1.1.46

os: win, apm: solo, release: 1.1.46
Downloading SITL from http://d3jdmgrrydviou.cloudfront.net/solo/sitl-win-v1.1.46.tar.gz
Traceback (most recent call last):
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Apps\WinPython-64bit-2.7.10.2\python-2.7.10.amd64\Scripts\dronekit-sitl.exe\__main__.py", line 9, in <module>
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\site-packages\dronekit\sitl\__init__.py", line 304, in main
    sitl.download(target, verbose=True)
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\site-packages\dronekit\sitl\__init__.py", line 119, in download
    return download(self.system, self.version, target, verbose=verbose)
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\site-packages\dronekit\sitl\__init__.py", line 97, in download
    testfile.retrieve(sitl_file, sitl_target + '/sitl.tar.gz')
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\urllib.py", line 364, in open_http
    return self.http_error(url, fp, errcode, errmsg, headers)
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\urllib.py", line 381, in http_error
    return self.http_error_default(url, fp, errcode, errmsg, headers)
  File "c:\apps\winpython-64bit-2.7.10.2\python-2.7.10.amd64\lib\urllib.py", line 386, in http_error_default
    raise IOError, ('http error', errcode, errmsg, headers)
IOError: ('http error', 404, 'Not Found', <httplib.HTTPMessage instance at 0x00000000029B2A08>)
```

hamishwillee, 2015-10-02:
I tried this again on Ubuntu running in vmware. ``dronekit-sitl solo-1.1.46`` gives similar error. The other builds seem to work fine, and I can connect to the SITL from Windows on the VMs IP address.

tcr3dr, 2015-10-15:
@hamishwillee Would you pull now from master, `dronekit-sitl --list` and try the available options? I've rebuilt the problematic ones and built new `copter-3.3` and `solo-1.2.0` binaries.

hamishwillee, 2015-10-15:
We will need plane 3.4.0 - just got released as current official build.

hamishwillee, 2015-10-15:
@tcr3dr On windows, fresh build from master of dk-sitl (fetches dk b6):

```
C:\Users\hamis>dronekit-sitl solo-1.2.0
os: win, apm: solo, release: 1.2.0
Downloading SITL from http://d3jdmgrrydviou.cloudfront.net/solo/sitl-win-v1.2.0.tar.gz
Extracted.
Note: Starting pysim for legacy SITL.
Pysim: C:\Python27\python2.7.exe C:\Python27\lib\site-packages\dronekit_sitl\pysim/sim_wrapper.py --simin=127.0.0.1:5502 --simout=127.0.0.1:5501 --fgout=127.0.0.1:5503 --home=-35.363261,149.165230,584,353 --frame=+
Execute: .\ArduCopter.elf
Simulating for frame +
Starting at lat=-35.363261 lon=149.165230 alt=584.0 heading=353.0
(error(10061, 'No connection could be made because the target machine actively refused it'), 'sleeping')
(error(10061, 'No connection could be made because the target machine actively refused it'), 'sleeping')
Traceback (most recent call last):
  File "C:\Python27\Scripts\dronekit-sitl-script.py", line 9, in <module>
    load_entry_point('dronekit-sitl==2.4.2', 'console_scripts', 'dronekit-sitl')()
  File "C:\Python27\lib\site-packages\dronekit_sitl\__init__.py", line 397, in main
    sitl.launch(args, verbose=True, local=local)
  File "C:\Python27\lib\site-packages\dronekit_sitl\__init__.py", line 236, in launch
    vehicle = dronekit.connect('tcp:127.0.0.1:5760')
  File "C:\Python27\lib\site-packages\dronekit\__init__.py", line 591, in connect
    state = MPFakeState(mavutil.mavlink_connection(ip), vehicle_class=vehicle_class)
  File "C:\Python27\lib\site-packages\pymavlink\mavutil.py", line 1067, in mavlink_connection
    return mavtcp(device[4:], source_system=source_system, retries=retries, use_native=use_native)
  File "C:\Python27\lib\site-packages\pymavlink\mavutil.py", line 863, in __init__
    self.port.connect(self.destination_addr)
  File "C:\Python27\lib\socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 10061] No connection could be made because the target machine actively refused it
```

hamishwillee, 2015-10-15:
For plane I connected to 127.0.0.1:5760, could connect and download parameters. What I noticed in several tests was that when I first start SITL the parameters can take a very long time to download - ie it can get stuck for two minutes on a particular parameter then proceed as normal. If I then disconnect and reconnect then connection/download of all parameters is "as normal" - about 30s.

So I think perhaps SITL is not quite ready when it says it is ready.
```
C:\Users\hamis>dronekit-sitl plane-3.3.0
os: win, apm: plane, release: 3.3.0
Downloading SITL from http://d3jdmgrrydviou.cloudfront.net/plane/sitl-win-v3.3.0.tar.gz
Extracted.
Execute: .\ArduPlane.elf --home=-35.363261,149.165230,584,353 --model=quad
Started model quad at -35.363261,149.165230,584,353 at speed 1.0
bind port 5760 for 0
Starting sketch 'ArduPlane'
Starting SITL input
Waiting for connection ....
Serial port 0 on TCP port 5760
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
```

hamishwillee, 2015-10-15:
Copter 3.2.1 - similar symptoms to Solo buiild above except that restarting it again didn't change behaviour.
```
C:\Users\hamis>dronekit-sitl copter-3.2.1
os: win, apm: copter, release: 3.2.1
Downloading SITL from http://d3jdmgrrydviou.cloudfront.net/copter/sitl-win-v3.2.1.tar.gz
Extracted.
Note: Starting pysim for legacy SITL.
Pysim: C:\Python27\python2.7.exe C:\Python27\lib\site-packages\dronekit_sitl\pysim/sim_wrapper.py --simin=127.0.0.1:5502 --simout=127.0.0.1:5501 --fgout=127.0.0.1:5503 --home=-35.363261,149.165230,584,353
Execute: .\ArduCopter.elf
Simulating for frame +
Starting at lat=-35.363261 lon=149.165230 alt=584.0 heading=353.0
(error(10061, 'No connection could be made because the target machine actively refused it'), 'sleeping')
(error(10061, 'No connection could be made because the target machine actively refused it'), 'sleeping')
Traceback (most recent call last):
  File "C:\Python27\Scripts\dronekit-sitl-script.py", line 9, in <module>
    load_entry_point('dronekit-sitl==2.4.2', 'console_scripts', 'dronekit-sitl')()
  File "C:\Python27\lib\site-packages\dronekit_sitl\__init__.py", line 397, in main
    sitl.launch(args, verbose=True, local=local)
  File "C:\Python27\lib\site-packages\dronekit_sitl\__init__.py", line 236, in launch
    vehicle = dronekit.connect('tcp:127.0.0.1:5760')
  File "C:\Python27\lib\site-packages\dronekit\__init__.py", line 591, in connect
    state = MPFakeState(mavutil.mavlink_connection(ip), vehicle_class=vehicle_class)
  File "C:\Python27\lib\site-packages\pymavlink\mavutil.py", line 1067, in mavlink_connection
    return mavtcp(device[4:], source_system=source_system, retries=retries, use_native=use_native)
  File "C:\Python27\lib\site-packages\pymavlink\mavutil.py", line 863, in __init__
    self.port.connect(self.destination_addr)
  File "C:\Python27\lib\socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 10061] No connection could be made because the target machine actively refused it
```
Looks like this build might be a meth addict :-0
```
  File "C:\Python27\lib\socket.py", line 224, in meth
```

hamishwillee, 2015-10-15:
Copter 3.3 worked great when connected to tcp:127.0.0.1:5760. Unlike plane build above parameters seemed to download reasonably first time.

hamishwillee, 2015-10-15:
Rover-2.50 connected to tcp:127.0.0.1:5760 and had same first-connect slow param download behaviors as the plane build. This build had "bad gyro health". @tcr3dr this still doesn't load good parameters does it?

tcr3dr, 2015-10-20:
To simplify, I'm going to integrate #48 first before addressing some of these (as they are race conditions).

hamishwillee, 2015-10-20:
@tcr3dr Excellent. Looking forward to actually being able to use this as my simulator-of-choice one day :-)