dronekit/dronekit-sitl #48: Package default eeprom with builds

tcr3dr, 2015-10-15:
Rather than have a parameter list, it is cheaper and more consistent to package a default eeprom file with the SITL runner.

hamishwillee, 2015-10-15:
Really? As long as this allows you to have a different eeprom for every vehicle/build then sure.

tcr3dr, 2015-10-20:
@hamishwillee Yeah. The alternative is very invasive, connecting then disconnecting to the created TCP port, leading to race conditions... no good!

tcr3dr, 2015-10-20:
Working in https://github.com/dronekit/dronekit-sitl/pull/50

hamishwillee, 2015-10-20:
Works for all builds? Ie separate/appropriate parameters/eeprom for copter vs plane and Rover and individual builds?

hamishwillee, 2016-02-08:
Closing. Looking at the way the eeprom is created it doesn't appear to have any vehicle/platform dependency.

closed 2016-02-08 by hamishwillee