dronekit/dronekit-sitl #53: Not loading default params

squilter, 2015-10-23:
I just followed the readme, and it didn't work.

`dronekit-sitl copter-3.3-rc5 -I0 -S --model quad --home=-35.363261,149.165230,584,353`

It would wipe the params, but not load the default ones.  So I couldn't arm because the RC wasn't calibrated.

tcr3dr, 2015-10-23:
@squilter Try a clean install, then `dronekit-sitl copter-3.3`.

tcr3dr, 2015-10-23:
@squilter (Also available on pip as dronekit-sitl==3.0.0)

squilter, 2015-10-23:
I ran `sudo pip2 install dronekit-sitl --upgrade`
then
`dronekit-sitl copter-3.3`
and now I get
`Exception: SITL binary /home/squilter/.dronekit/sitl/copter-3.3/apm does not exist.`

squilter, 2015-10-23:
can confirm that I now have `dronekit-sitl==3.0.0`

tcr3dr, 2015-10-23:
Can you run `tree /home/squilter/.dronekit` for me? Did it mention the binary downloading or not?

squilter, 2015-10-23:
It used to mention downloading binary.  Not anymore.

```
/home/squilter/.dronekit
└── sitl
    ├── copter-3.2.1
    │   ├── ArduCopter.elf
    │   ├── eeprom.bin
    │   └── logs
    ├── copter-3.3
    │   ├── ArduCopter.elf
    │   ├── logs
    │   └── terrain
    │       └── S36E149.DAT
    ├── copter-3.3-rc5
    │   ├── ArduCopter.elf
    │   ├── logs
    │   └── terrain
    │       └── S36E149.DAT
    └── sitl.tar.gz

9 directories, 7 files
```

tcr3dr, 2015-10-23:
Ah. `setup.py` is now actually just blowing away the `~/.dronekit/sitl` directory on reinstall (since all the earlier binaries were pretty inconsistent, and the directory shouldn't store state). Feel free to do the same, or manually remove the copter-3.3 directory.

tcr3dr, 2015-10-23:
I think I can handle this edge case automatically.

squilter, 2015-10-23:
All is working now.  Thanks.

closed 2015-10-23 by squilter

tcr3dr, 2015-10-23:
@squilter Great. Can give this a try if you'd like. Probably will just republish with it. https://github.com/dronekit/dronekit-sitl/pull/55

squilter, 2015-10-23:
Not sure how to downgrade with pip, so I don't know how to test it.

tcr3dr, 2015-10-23:
Mmmm we'll just assume it works.

Software engineering.

hamishwillee, 2015-10-23:
No wonder I feel like 50% of my job is QA .... :-)

tcr3dr, 2015-11-07:
@hamishwillee Many Q, some A ;)