dronekit/dronekit-sitl #56: No module named tools

franknitty69, 2015-11-03:
when issuing any Dronekit-sitl command such as dronekit-sitl --list, dronekit/lib/__init__.py fails trying to import errprinter from dronekit.tools.

OS: OSX
dronekit-sitl version: 3.0.1

**steps to reproduce**
1. install dronekit-sitl
2. issue any dronekit-sitl command, i.e. **dronekit-sitl --list**

**result**
File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/dronekit/lib/__init__.py", line 7, in <module>
    from dronekit.tools import errprinter
ImportError: No module named tools

**expected result**
dronekit-sitl commands should work as described

hamishwillee, 2015-11-04:
This is actually (probably) a Dronekit issue. Fix being looked at in https://github.com/dronekit/dronekit-python/pull/409

franknitty69, 2015-11-04:
resolved per https://github.com/dronekit/dronekit-python/pull/409

closed 2015-11-04 by franknitty69