dronekit/dronekit-sitl #60: APM Params are not preserved

wsilva32, 2015-12-16:
APM params, such was WPNAV_SPEED, are not being preserved between dronekit-sitl launches.

tcr3dr, 2015-12-21:
We want a way to set the eeprom to a local file, i.e. `dronekit-sitl --eeprom=my.eeprom` so that you can preserve state between launches.

This should be practical by syncing the file generated in the temp SITL directory (mav.eeprom?) to the destined file, though we'll need to sync the file over the life of the program.

Is there a file change watcher in Python we can leverage?

Chambana, 2016-02-07:
@tcr3dr  It seems like my virtual rangefinder isn't getting started as part of my SITL.  Would that be tied to the above issue?  

I'm setting the following params using MAVProxy communicating with the SITL on OSX, using the instructions at http://dev.ardupilot.com/wiki/using-sitl-for-ardupilot-testing/#adding_a_virtual_rangefinder: 
param set SIM_SONAR_SCALE 10
param set RNGFND_TYPE 1
param set RNGFND_SCALING 10
param set RNGFND_PIN 0
param set RNGFND_MAX_CM 5000

and using "dronekit-sitl copter" to re-start the SITL after setting those params. 

After I reset the SITL, no rangerfinder messages appear in the MAVLink inspector.  Is there another way to enable a virtual rangefinder with dronekit SITL?

hamishwillee, 2016-02-07:
@Chambana If you look in the SITL download directory (on my computer C:\Users\hamis\.dronekit\sitl\copter-3.3) you will see `default.parm` and `default_eeprom.bin`. I would TRY seeing if updating the the default.parm does what you want. If not, I would try deleting/renaming default_eeprom.bin and then trying again (might not work!). 

We do need a clear answer to this use case, so I will discuss with @peterbarker.

hamishwillee, 2016-02-08:
@tcr3dr @mrpollo 

@peterbarker and I did a little bit of testing and digging and the behaviour is a bit confusing. What we think happens is that as soon as SITL starts dk-sitl copies the `default_eeprom.bin` into a working directory, and this working directory version is what is loaded by SITL. The result is that any changes made to the parameters between versions will be cleared. I also did a check and the `default.parms` aren't loaded (so you can't control behaviour by modifying those). Is that all correct?

If we are correct then rather than this suggestion (https://github.com/dronekit/dronekit-sitl/issues/60#issuecomment-166400723) how about we provide a flag `-k` / `-keep` to SITL to disable the copying of `default_eeprom.bin` on SITL start. You would just specify this flag to keep any changes you made, but otherwise you'd have the same behaviour.