dronekit/dronekit-sitl #62: APM Planner 2.0 disconnects

alduxvm, 2015-12-30:
I'm using  `dronekit-sitl copter ` to start my sitl. 

Then in another terminal I do:   `mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551 ` 

Then I open APM Planner 2.0 (version 2.0.18) and it connects to the vehicle, but after like 10-15 seconds the voice says "Link lost to system 1"... Only if I restart the application I can see updates... but only lasts for 10-15 seconds. 

Is there something I need to do to prevent those disconnects?

mrpollo, 2015-12-30:
What OS are you running on?

alduxvm, 2015-12-30:
OS X El Capitan

mrpollo, 2015-12-30:
Having been able to replicate on my local using 10.11 latest, can you verify you have the latest dronekit-sitl and mavproxy?

```
pip install dronekit-sitl --upgrade
pip install mavproxy --upgrade
```

alduxvm, 2015-12-30:
Ok, I install them today... but anyhow I did it again, and this is the result:

`Successfully installed dronekit-sitl-3.0.1`
`Successfully installed mavproxy-1.4.38`

And it remains the same, after 13 seconds... "Link lost with system 1"

Should I remove planner 2 and install again?

mrpollo, 2015-12-30:
just tried this again with latest APM Planner, still no success, can you set `set moddebug 3` on mavproxy it might give us more to work on

alduxvm, 2015-12-30:
I did a video-screenshot: https://altax.net/videos/dronekit-sitl.mp4

mrpollo, 2015-12-30:
Ah, you are arming!

Steps to replicate:
1. launch `dronekit-sitl copter`
2. launch mavproxy `mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551`
3. open **APM Planner 2.0.app** (OS X)
4. on mavproxy.py run `arm throttle`

mrpollo, 2015-12-30:
just confirmed this happens on `sitl-osx-copter-3.3` no need to have **APM Planner 2.0.app** running, will investigate

alduxvm, 2015-12-30:
but my command is:

`dronekit-sitl copter -IO --model quad --home=55.870592,-4.287598,0,0`

for 3.3 is not? `dronekit-sitl copter-3.3 -IO --model quad --home=55.870592,-4.287598,0,0` 

unless 3.3 is the default...

mrpollo, 2015-12-30:
`copter` defaults to latest, in this case `copter-3.3`

alduxvm, 2015-12-30:
ok, so, which version should I use to make it work properly?

mrpollo, 2015-12-30:
not sure what the problem is yet, you might get better luck with a different copter version, please let me know if that works

alduxvm, 2015-12-30:
ok, which version can I use?

When I do: `dronekit-sitl copter-3.2` it give me this error:

`
Downloading SITL from http://dronekit-assets.s3.amazonaws.com/sitl/copter/sitl-osx-copter-3.2.tar.gz
Traceback (most recent call last):
  File "/usr/local/bin/dronekit-sitl", line 9, in <module>
    load_entry_point('dronekit-sitl==3.0.1', 'console_scripts', 'dronekit-sitl')()
  File "/usr/local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 415, in main
    sitl.download(system, version, target=target, verbose=True)
  File "/usr/local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 136, in download
    return download(system, version, target, verbose=verbose)
  File "/usr/local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 107, in download
    testfile.retrieve(sitl_file, sitl_target + '/sitl.tar.gz')
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 364, in open_http
    return self.http_error(url, fp, errcode, errmsg, headers)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 381, in http_error
    return self.http_error_default(url, fp, errcode, errmsg, headers)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 386, in http_error_default
    raise IOError, ('http error', errcode, errmsg, headers)
IOError: ('http error', 404, 'Not Found', <httplib.HTTPMessage instance at 0x10db86f38>)
`

hamishwillee, 2015-12-30:
@mrpollo @alduxvm At the moment there is only one version of each of the vehicles (Copter 3.2.1 kept on getting exceptions on windows so we removed it for more testing). 

Does the link work if you turn off MAVProxy and connect directly to TCP port 5670 from APM planner?

alduxvm, 2015-12-31:
that worked @hamishwillee !!!

1.- Start sitl like this: `dronekit-sitl copter -IO --model quad`
2.- Open "APM Planner 2.0.app (OS X)" and add a TCP link @ localhost and port 5760
3.- Start mavproxy with this: `mavproxy.py --master tcp:127.0.0.1:5763`

And after that, there is not "Link lost" and I can do whatever I want inside mayproxy.

My guess there is a problem with the UDP `--out` of mavproxy?? anyhow, I can do tests now using this solution at the moment.

mrpollo, 2015-12-31:
thanks for posting your solution @alduxvm, we'll get to the bottom of this