dronekit/dronekit-sitl #63: SITL won't load custom firmware on OSX, permission denied spawning subprocess

Chambana, 2016-01-04:
I can load the stock firmware (Copter 3.3) fine on OSX El Capitan, but attempts to load local custom firmware fails on launch.  

This:
sudo dronekit-sitl ./ArduCopter-v2.px4

Throws the following exception:
os: osx, local binary: /Users/______/ardupilot/ArduCopter/ArduCopter-v2.px4
Note: Starting pysim for legacy SITL.
Pysim: /usr/local/opt/python/bin/python2.7 /usr/local/lib/python2.7/site-packages/dronekit_sitl/pysim/sim_wrapper.py --simin=127.0.0.1:5502 --simout=127.0.0.1:5501 --fgout=127.0.0.1:5503 --home=-35.363261,149.165230,584,353 --frame=+
Execute: /Users/______/ardupilot/ArduCopter/ArduCopter-v2.px4 /Users/______/ardupilot/ArduCopter/ArduCopter-v2.px4
Traceback (most recent call last):
  File "/usr/local/bin/dronekit-sitl", line 9, in <module>
    load_entry_point('dronekit-sitl==3.0.1', 'console_scripts', 'dronekit-sitl')()
  File "/usr/local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 417, in main
    sitl.launch(args, verbose=True)
  File "/usr/local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 239, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 13] Permission denied

tcr3dr, 2016-01-04:
Hey @Chambana, can you run `chmod +x` on `./ArduCopter-v2.px4` and check if the issue still exists?

Also, can you run `ls -la ./ArduCopter-v2.px4` and paste the output of that in a reply to this comment?

Chambana, 2016-01-05:
Here are the results of ls -la ./ArduCopter-v2.px4
     -rw-r--r--  1 root  staff  780997 Jan  2 23:59 ./ArduCopter-v2.px4

I failed to realize that the file's permissions weren't set correctly. That said, after running chmod +x I still have this issue:

dronekit-sitl ./ArduCopter-v2.px4
os: osx, local binary: ./ArduCopter-v2.px4
Note: Starting pysim for legacy SITL.
Pysim: /usr/local/opt/python/bin/python2.7 /usr/local/lib/python2.7/site-packages/dronekit_sitl/pysim/sim_wrapper.py --simin=127.0.0.1:5502 --simout=127.0.0.1:5501 --fgout=127.0.0.1:5503 --home=-35.363261,149.165230,584,353 --frame=+
Execute: /Users/_____/ardupilot/ArduCopter/ArduCopter-v2.px4 /Users/_____/ardupilot/ArduCopter/ArduCopter-v2.px4
Traceback (most recent call last):
  File "/usr/local/bin/dronekit-sitl", line 9, in <module>
    load_entry_point('dronekit-sitl==3.0.1', 'console_scripts', 'dronekit-sitl')()
  File "/usr/local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 417, in main
    sitl.launch(args, verbose=True)
  File "/usr/local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 239, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error

hamishwillee, 2016-01-05:
I could be wrong, but isn't the file that needs to be run ArduCopter.elf?

Chambana, 2016-01-05:
Hmmm.... when you compile the firmware, the last output statement from the log is "PX4 ArduCopter Firmware is in ArduCopter-v2.px4"

hamishwillee, 2016-01-05:
@Chambana Building following these instructions? http://dev.ardupilot.com/wiki/setting-up-sitl-on-linux/

Chambana, 2016-02-19:
@hamishwillee  I'm using dronekit-sitl, so I don't know if the vagrant guide would apply as much.  In any case, I installed dronekit-sitl using pip and it works great when executing dronekit-sitl with the included ardupilot firmware binaries from the OSX terminal.  When I load custom firmware that I compiled locally, dronekit-sitl fails to launch on the call to popen.  I tried launching dronekit-sitil with both the px4io-v2.bin and the px4io-v2.elf files that get built from the ardupilot source using "make px4-v2".  I get "Exec format error"....

  File "/usr/local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 240, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error

hamishwillee, 2016-02-19:
Can you please run “file ArduCopter-v2.px4” and let us see the results.That is probably a binary image, not an executable.

Chambana, 2016-02-24:
Sure.  The output of "file ArduCopter-v2.px4" is: 

ArduCopter-v2.px4: ASCII English text, with very long lines

hamishwillee, 2016-02-24:
I am 99% sure that is a binary image and not a sitl executable. An executable will be something like .elf. You need to build for a SITL target.

closed 2016-04-02 by Chambana