dronekit/dronekit-sitl #67: update readme with python3 support

mrpollo, 2016-01-11:
we now support python3, we either remove our `pip2` traces and let the user decide, or add instructions for `pip3`

hamishwillee, 2016-01-19:
Hi @mrpollo 

Never used Pip3. Internet seems to indicate that pip, pip2, pip3 are just different revisions of the same program, which presumably in version pip3 knows to get Python 3 versions of the binaries. Some of the other resources out there seem to indicate that if you install pip3, then calling pip will get you that anyway - in other words, you get the pip that is appropriate for your current environment.

Is that correct? If so, then the following instructions should be python-version-agnostic right ? :
```
pip install dronekit-sitl -UI
```

mrpollo, 2016-01-19:
thats correct!, i would much rather go with just `pip` instead of `pip2`, `pip3`

hamishwillee, 2016-01-19:
@mrpollo Agreed. Only issue is that this doesn't work for me on Windows. So action is to remove the "2" in pip2 commands in readme.md, but only after #68 is addressed.

cclauss, 2016-10-02:
On Mac OSX we use [homebrew](http://brew.sh) to install both Python2 and Python3 (and Pypy and Pypy3) on the same machine.  The use of pip2 and pip3 (and pip_pypy and pip_pypy3) to enable users to install a module for one particular Python implementation and not for the others.  I agree with @mrpollo above that each user needs to understand how pip is setup on their own machines.