dronekit/dronekit-sitl #68: Error installing dronekit-sitl into python 3 environment on Windows

hamishwillee, 2016-01-19:
See debug output below.
```
C:\Program Files>pip list
pip (7.1.2)
setuptools (18.2)

C:\Program Files>pip install dronekit-sitl -UI
Collecting dronekit-sitl
  Using cached dronekit_sitl-3.0.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 20, in <module>
      File "C:\Users\hamis\AppData\Local\Temp\pip-build-u2ghls8d\dronekit-sitl\setup.py", line 32
        print 'Cleared cached SITL binaries.'
                                            ^
    SyntaxError: Missing parentheses in call to 'print'

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in C:\Users\hamis\AppData\Local\Temp\pip-build-u2ghls8d\dronekit-sitl
```

bmann33, 2016-04-13:
Hello, I am new to this language and I am trying to install the samething on my computers. Running windows 10 64 bit. I get the same error. That error 1. And when i try to install it using the pip install github method, I get error [WinError 2]. Is there any fixes to this solution? I desperately need dronekit-sitl to run, Thanks!

Microsoft Windows [Version 10.0.10586]
(c) 2015 Microsoft Corporation. All rights reserved.

C:\Users\Brian Mann>pip list
dronekit (2.4.0)
monotonic (0.6)
pip (8.1.1)
pymavlink (1.1.72)
requests (2.9.1)
requests-oauth (0.4.1)
setuptools (20.7.0)

C:\Users\Brian Mann>pip install dronekit-sitl
Collecting dronekit-sitl
  Using cached dronekit_sitl-3.0.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\BRIANM~1\AppData\Local\Temp\pip-build-8z0otk_t\dronekit-sitl\setup.py", line 32
        print 'Cleared cached SITL binaries.'
                                            ^
    SyntaxError: Missing parentheses in call to 'print'

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in C:\Users\BRIANM~1\AppData\Local\Temp\pip-build-8z0otk_t\dronekit-sitl\

C:\Users\Brian Mann>pip install git+https://github.com/dronekit/dronekit-sitl
Collecting git+https://github.com/dronekit/dronekit-sitl
  Cloning https://github.com/dronekit/dronekit-sitl to c:\users\brianm~1\appdata\local\temp\pip-71m24lw2-build
  Error [WinError 2] The system cannot find the file specified while executing command git clone -q https://github.com/dronekit/dronekit-sitl C:\Users\BRIANM~1\AppData\Local\Temp\pip-71m24lw2-build
Cannot find command 'git'

C:\Users\Brian Mann>

cerv0077, 2016-04-13:
I have the same problem. Anyone to help us please??

bmann33, 2016-04-13:
Hey cerv0077, i fixed my issue. What i did was uninstalled everything and started fresh with a lower version of python. (2.7 to be exact). I originally had the latest one, which was 3.5 or something like that. Hope this helps

hamishwillee, 2016-04-14:
@peterbarker Was this fixed in master - if so, can you do a new release?