dronekit/dronekit-sitl #71: Error executing dronekit-sitl in Raspberry Pi

sbneves, 2016-02-03:
I am trying to run dronekit-sitl on Raspberry Pi, without success. I have already run it on linux normally.
Here is what happens:

_pi@raspberrypi:_~ $ dronekit-sitl copter-3.3
os: linux, apm: copter, release: 3_.3
SITL already Downloaded.
Execute: /home/pi/.dronekit/sitl/copter-3.3/apm --home=-35.363261,149.165230,584,353 --model=quad
Traceback (most recent call last):
  File "/usr/local/bin/dronekit-sitl", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 417, in main
    sitl.launch(args, verbose=True)
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 239, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error_

Can anyone help me?

hamishwillee, 2016-02-04:
@sbneves RPi isn't a supported platform for DK-sitl so I doubt the dev team will take much interest in this. My suggestion is to[ask google](https://www.google.com.au/search?q=RPi+%5BErrno+8%5D+Exec+format+error&ie=utf-8&oe=utf-8&gws_rd=cr,ssl&ei=iZeyVq33B8S90gTBj7fADw).

For my interest, why do you want to run an autopilot simulation on an RPi?

manik9000, 2016-02-09:
I am having same problem on Uuntu 14.04 LTE
any solution?

peterbarker, 2016-02-09:
@manik9000 What architecture is your Ubuntu machine?

Up until your comment I was assuming the problem was simply that the downloaded ArduCopter.elf was compiled for the x86 architecture, whereas the Pi is an ARM architecture.  We only provide binaries for Windows and Linux/x86.

@sbneves At the very least we need to make that error clearer.  "Exec format error" is the sort of error  would expect when attempting to run an x86 binary on an ARM machine.

manik9000, 2016-02-09:
@peterbarker : it's ubuntu 14.04/x86, Intel® Pentium(R) CPU P6200 @ 2.13GHz × 2

manik9000, 2016-02-09:
@peterbarker [solved]

so issue was with `/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py` file
at line # 239
so to resolve this problem I just changed 
`p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)`
to
`p = Popen([self.path] + args, cwd=wd, shell=True, stdout=PIPE, stderr=PIPE)`

I hope this post will be helpful for someone else.

manik9000, 2016-02-09:
but now there is another problem, 
as soon as I made those changes and executed `dronekit-sitl copter` 
it s throwing error `..... /copter-3.3/apm: 1: /home/thinkerbox/.dronekit/sitl/copter-3.3/apm: Syntax error: "(" unexpected`

where as 'dronekit-sietl -h' is working fine

any help?

mangritosh, 2016-02-10:
I have exactly the same problem on 14.04 Ubuntu x86.

hamishwillee, 2016-02-11:
@peterbarker Did you do anything to address your comment https://github.com/dronekit/dronekit-sitl/issues/71#issuecomment-181659412 (e.g. create a PR or issue?)

>At the very least we need to make that error clearer. "Exec format error" is the sort of error would expect when attempting to run an x86 binary on an ARM machine

@manik9000 @mangritosh I'm running this without change on Ubuntu 14.04 LTS running in a VM. This only has Python 2 installed. Did you install the very latest version?
```
sudo pip install git+https://github.com/dronekit/dronekit-sitl -UI
```

@mangritosh can you please paste in what you see (basically I want to confirm that it is exactly the same problem)

manik9000, 2016-02-11:
@hamishwillee I used `sudo pip install dronekit-sitl -UI`
I guess both the commands are same, isn't it?

mangritosh, 2016-02-11:
@peterbarker This is the output:

`Traceback (most recent call last):
  File "/usr/local/bin/dronekit-sitl", line 9, in <module>
    load_entry_point('dronekit-sitl==3.0.1', 'console_scripts', 'dronekit-sitl')()
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 417, in main
    sitl.launch(args, verbose=True)
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 239, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error
`

hamishwillee, 2016-02-12:
@manik9000 

> @hamishwillee I used sudo pip install dronekit-sitl -UI
> I guess both the commands are same, isn't it?

They are only the same if a version of the package on PyPi is created whenever we update dronekit-sitl. I don't know if that is the case, or think it is the problem, but when debugging it is nice to be able to say that we're running the same environment

hamishwillee, 2016-02-12:
@mangritosh @mangritosh I can't reproduce this error on Ubuntu. I've tried both Python2 and Python3 and everything works fine. 

> so issue was with /usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py file
> at line # 239
> so to resolve this problem I just changed
> p = Popen([self.path] + args, cwd=wd, [...]

Can you please try `shell=False`? From [the Python docs](https://docs.python.org/3/library/subprocess.html#subprocess.Popen) it sounds like this should always be false (even on windows). 

Can you also try this with both Python2 and Python3? I think you can change to Python 3 using the command:
```
alias python=python3
```

manik9000, 2016-02-12:
@hamishwillee 
Thanks
I will try to install it again as you suggested and will let  you know how it goes..

MDB22, 2016-02-13:
@hamishwillee I am in the same situation as OP, attempting to use DK-sitl on a Raspberry Pi running Raspbian with Python 2.7. Would you suggest using Ubuntu on a computer instead? 

The RPi is intended to be an on-board companion computer, so I figured testing our code on the RPi was a good idea, but correct me if I'm wrong.

I've pulled the latest version from the repository, and have tried all the suggestions here: 
> p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
> to
> p = Popen([self.path] + args, cwd=wd, shell=True, stdout=PIPE, stderr=PIPE)

results in

> ..... /copter-3.3/apm: 1: /home/thinkerbox/.dronekit/sitl/copter-3.3/apm: Syntax error: "(" unexpected
as above

I've also tried
> p = Popen([self.path] + args, cwd=wd, shell=False, stdout=PIPE, stderr=PIPE)

as you suggested, and continue to get the "Exec format error".

hamishwillee, 2016-02-14:
@MDB22 Testing your DroneKit-Python code running on RPi makes sense because that is where you'll want to run your scripts. However also running DK-SITL on that same platform isn't necessary and won't work (we don't supply ARM binaries). What you need to do for simulation is have the SITL running on another machine on your network and make your scripts connect to it's network address (the actual fully qualified address of course, not the loopback you might use if running on the same machine).

hamishwillee, 2016-02-14:
@manik9000 Another thing to try is modify the shebang at the top of the page from:
```
#!/usr/bin/python
```
to (try a few?)
```
#!/usr/bin/env
```
or
```
#!/usr/bin/env python
```
or
```
#!/usr/bin/env python3
```

MDB22, 2016-02-16:
@hamishwillee thanks for your help, successfully up and running with SITL on Windows, communicating to Raspberry Pi :)

hamishwillee, 2016-02-16:
@MDB22 Happy days :-) Can you tell me what you had to change on windows (exactly), if anything to get this to work.

MDB22, 2016-02-16:
After installing DrokeKit-SITL on my Windows machine and DroneKit on the Raspberry Pi, I executed dronekit-sitl from the Windows command line, where it waits for a connection on port 5760.

On the RPi I first tested with MAVProxy:
`mavproxy.py --master tcp:windows.device.ip.address:5760 --out windows.gcs.ip.address:14550`

I can then connect my GCS (Mission Planner) to UDP port 14550 and it receives telemetry from the SITL simulation, and I can issue MAVLink flight commands from the GCS to the simulation.

I also tested using parts of the "Quick Start" guide: [http://python.dronekit.io/guide/quick_start.html](url)

SITL setup and launch is handled on the Windows machine, and connecting from the Pi is as simple as replacing
`vehicle = connect('tcp:127.0.0.1:5760', wait_ready=True)`
with
`vehicle = connect('tcp:windows.device.ip.address:5760', wait_ready=True)`
and everything proceeds as normal.

hamishwillee, 2016-02-16:
Thanks very much. That is pretty much the "standard setup". Glad you got it working.

hamishwillee, 2016-02-16:
I am going to close this issue since the RaPi stuff is sorted.

@manik9000 Did you get this working? If not, can you please create another issue for this.

closed 2016-02-16 by hamishwillee

Turbullance, 2020-11-12:
> @peterbarker [solved]
>
> so issue was with `/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py` file
> at line # 239
> so to resolve this problem I just changed
> `p = Popen([self.path] + [...]

I tried but i couldnt save it as it states it isnt writeable