dronekit/dronekit-sitl #77: Add simplest possible API for launching sitl

hamishwillee, 2016-02-11:
The DK python examples have to have a lot of "boilerplate" for launching examples.:
```
from dronekit_sitl import SITL
     sitl = SITL()
     sitl.download('copter', '3.3', verbose=True)
     sitl_args = ['-I0', '--model', 'quad', '--home=-35.363261,149.165230,584,353']
     sitl.launch(sitl_args, await_ready=True, restart=True)
```
It would be great if we could have a single command that downloads the latest version by default (if needed) and then launches it ... mirroring the normal command usage.

So perhaps add a method like this:
```
sitl, connection_string=SITL.launch_default(vehicle=copter, version=latest,verbose=True,args="thestuff",await_ready=True)
```
where you could specify everything if you like, but otherwise it just downloads and launches the latest copter. 

- This should be static, so that you don't need to create an instance of SITL to use it. 
- this should return an instance of sitl that you can use to clean up the vehicle later
- connection_string is the port/url you need to connect to this SITL instance.

The boilerplate would then become:
``` 
from dronekit_sitl import SITL
sitl, connection_string=SITL.launch_default()
```

Bonus points if you can work out how to auto destroy the sitl instance "when it is no longer needed/when the script exits".

Open to discussion on other ways to implement this sort of functionality. Considered creating a utility function in DK-python for this but I think it is better here.

peterbarker, 2016-02-11:
https://github.com/dronekit/dronekit-python/blob/flight-replay-localfile/examples/flight_replay/flight_replay.py#L24

hamishwillee, 2016-02-11:
@peterbarker Yes, but the boilerplate is still there, you've just put it in a function. Slightly better :-)

peterbarker, 2016-05-14:
https://github.com/dronekit/dronekit-python/pull/624
https://github.com/dronekit/dronekit-sitl/pull/87

peterbarker, 2016-05-24:
The commit to address this was merged.

closed 2016-05-24 by peterbarker