dronekit/dronekit-sitl #79: Plane 3.3.0 binaries are unflyable on Windows and Linux

hamishwillee, 2016-02-24:
If you do ``dronekit-sitl plane`` the plane will start but essentially is completely unstable - it will spin around, spit out bad altitudes etc. Needs to be rebuilt with working parameters.

hamishwillee, 2016-02-24:
@peterbarker Another one for our list ....

ianamason, 2016-11-13:
Any progress on this?

hamishwillee, 2016-11-13:
Sorry, no. @peterbarker Have you ever built the binaries? Would be good to automate this a bit.

jakebeal, 2017-03-17:
This does not work on OS X either. It may be linked to the fact that dronekit-sitl plane launches with a "quad" model, and there is no "plane" model available.

al-mahi, 2019-07-01:
sitl = SITL(path=~/ardupilot/build/sitl/bin/arduplane) worked for me.
You have to install ardupilot and build. Its a workaround but not a solution.