dronekit/dronekit-sitl #80: Updating the Solo firmware version

DarylWM, 2016-02-29:
I'm wondering whether I can use it with a more recent version of the Solo firmware (1.2.23) than what's currently available in the --list (1.2.8) for OS X. Happy to build it myself if there are some instructions to follow.

hamishwillee, 2016-02-29:
NOTE: I have not yet tried this!

The script for building and instructions are here: https://github.com/dronekit/dronekit-sitl/tree/master/stage

This would have to be run in the environment you normally build SITL in natively - e.g. Cygwin for windows.

I'm keen to find out how you get one since at some point soon I do want to create some new binaries.

Is this sufficient?

DarylWM, 2016-03-01:
Thanks @hamishwillee. I tried this and build.sh fails on line 113:
`source ./env/bin/activate`
because the $STARTDIR/env directory doesn't exist. Not sure yet why the script would think it does.

hamishwillee, 2016-03-01:
The script does:
```
STARTDIR=$(pwd)
```
As I understand it, that means that it is set as whatever directory you are in.

What environment are you running this in? What do you get if you type pwd on the bash shell?

DarylWM, 2016-03-04:
I'm on OS X 10.11.3. In the `/Users/daryl/dronekit-sitl/stage` directory, I ran `./build.sh solo 1.2.19`. It exits with the error `./build.sh: line 113: ./env/bin/activate: No such file or directory`. I'm no bash wizard but it seems the script is expecting a `./env` directory off the `stage` directory, where I'm running it.

DarylWM, 2016-03-04:
Ah, I see that I missed a few steps...
```
brew tap homebrew/versions
brew install gcc48 gawk
sudo easy_install pip
sudo pip install awscli virtualenv
python setup.py install
```
and then...

`./build.sh solo 1.2.19
`

bheiland, 2016-03-07:
Trying to follow your steps to build a more current version of solo firmware but I don't see where you are finding the latest solo files.

DarylWM, 2016-03-07:
I think the idea is to use the tagged releases https://github.com/tcr3dr/ardupilot-releases/releases