dronekit/dronekit-sitl #83: Can't Running SITL : 'dronekit-sitl copter' error

NiteShade, 2016-03-28:
Hello

I am trying to get the dronekit-sitl up and running and am running into errors. I've tried installing from pip : 

`pip install dronekit-sitl -UI`

but when I install  and try running `dronekit-sitl copter` I get the error:

```
skyhawk@ubuntu:~$ dronekit-sitl copter
os: linux, apm: copter, release: stable
SITL already Downloaded.
Execute: /home/skyhawk/.dronekit/sitl/copter-3.3/apm --home=-35.363261,149.165230,584,353 --model=quad
Traceback (most recent call last):
  File "/usr/local/bin/dronekit-sitl", line 9, in <module>
    load_entry_point('dronekit-sitl==3.0.1', 'console_scripts', 'dronekit-sitl')()
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 417, in main
    sitl.launch(args, verbose=True)
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 239, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error
```

I`m using linux ubuntu 14.04 
Any help would be appreciated

NiteShade, 2016-03-31:
I've been using ubuntu 14.04 (x86 architecture) running in VM , .... but  I used OS windows 7 64 bit, does it affect the error?

Chambana, 2016-04-02:
Same problem.  Running ArchLinux on ARM-based Raspberry Pi.  Dronekit-SITL installs via PIP successfully, but throws the "Exec Format Error":

[root@alarmpi /]# dronekit-sitl copter                                         
os: linux, apm: copter, release: stable
SITL already Downloaded.
Execute: /root/.dronekit/sitl/copter-3.3/apm --home=-35.363261,149.165230,584,353 --model=quad
Traceback (most recent call last):
  File "/usr/bin/dronekit-sitl", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 417, in main
    sitl.launch(args, verbose=True)
  File "/usr/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 239, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error

hamishwillee, 2016-04-02:
@Chambana Not supported - we don't supply ARM binaries, and even if we did, not sure why you'd want to simulate on RPi.

@NiteShade Running in VM on Windows 7 shouldn't affect anything ... Unfortunately I can't debug this. Again, suggest you try native build.

Chambana, 2016-04-04:
@hamishwillee  Thanks for clarifying -- I wasn't aware that ARM isn't supported in Dronekit-sitl. Perhaps that is documented somewhere in the Readme and I just missed it.

Re why I'd want to run DK SITL on a RPi...  My DroneKit project runs on a Raspberry Pi companion computer and uses RPi-specific hardware/binaries (e.g. Raspivid, GPIO, etc).  So, I have to run my project on the Pi itself as it would be a pain to simulate all that stuff on my x86 machine.  I think the only alternative I have is to hook up the real autopilot hardware to the RPi for testing or perhaps I can pipe the DK SITL connection from my x86 machine to the RPi.   Cheers.

hamishwillee, 2016-04-04:
@Chambana Yes, your best option is to talk to SITL hosted on another machine - which is easy if they can both be on the same IP network. Alternatively you might also be able to build SITL yourself and run it on the RaPi (though I have no idea if that is "supported")

NiteShade, 2016-04-08:
thank you so much @hamishwillee  

@hamishwillee  may i ask again ?
how to add tcp port (eg 5670, 5780, 5790) in dronekit-sitl ?

hamishwillee, 2016-04-10:
@NiteShade I believe you got pointed to this example https://github.com/peterbarker/dronekit-python/tree/source-system-filtering/examples/multivehicle [from gitter](https://gitter.im/dronekit/dronekit-python?at=570ad1a92a2f4d4276133fa4) . Please post separate issues for separate problems. Also, I am presuming that you're asking how to support multiple drones, not multiple connections to the same drone.

fnoop, 2016-04-17:
@Chambana - sitl runs perfectly on ARM for me on raspberry, odroid and beaglebone-black.  You just need to compile the ardupilot firmware on the raspberry (make -j4 linux) and then point sitl at that binary.

NathanAhmad, 2016-04-20:
@fnoop - Could you give a little more guidance on pointing DroneKit-SITL to the ardupilot compilation of SITL, please?  I'd really appreciate it.  

For background, I am also trying to run dronekit-sitl simulations on a RPi.  I'm getting the same "OSError: [Errno 8] Exec format error" as @NiteShade, so I installed the ardupilot SITL by following this:  http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html
Now I can get a simulated drone running on the Pi using sim_vehicle.sh, but I'm still getting the same error when I try to start a simulation in Python using DroneKit-Python and DroneKit-sitl.  

Thanks

fnoop, 2016-04-20:
@NathanAhmad sure, have you compiled ArduPilot on the RPi?  In the sitl link you've followed, where it says
```
git clone git://github.com/ArduPilot/ardupilot.git
```
You have to actually compile the firmware for generic linux, so:
```
cd ardupilot/ArduCopter
make -j4 sitl
```
(sorry made a typo previously, it's make sitl, not make linux, and the -j4 assumes you have a RPi2 with quad cores).
Then just point dronekit-sitl at the end firmware:
```
dronekit-sitl ardupilot/ArduCopter/ArduCopter.elf
```
Note that when you use the dronekit-python example scripts, you have to specify --connect to the sitl you've just fired up, otherwise it tries by default to download a precompiled ardupilot binary at which point you'll get the same exec error, so something like:
```
example-script.py --connect tcp:localhost:5760
```

NathanAhmad, 2016-04-20:
@fnoop - Thank you so much!  I've got everything working now.  I really appreciate the help 👍

fnoop, 2016-04-20:
Great :)

jainakshay91, 2016-04-23:
Hi,
I am using an odroid C1+ with Linux version 3.10.80-135. I am getting the same error as many other people on running: dronekit-sitl copter. Can anybody provide some guidance on this. 

@fnoop is your method specific to an RPi2 with quad core processor?

hamishwillee, 2016-04-24:
The error is because dk-sitl doesn't ship ARM binaries. The solution is to build your own natively. Glad you got it working.

fnoop, 2016-04-24:
@jainakshay91 that's interesting, I use -j8 on odroid xu4.   Great you got it working, viva la ARM!

jainakshay91, 2016-04-27:
I know this does not belong here, but still I am asking this in need for some assistance. I started SITL and just to test ran simple_goto.py. The drone connects and initializes but then gets stuck on Pre-Arm check: RC not calibrated. There was an issue raised two years ago: https://github.com/Dronecode/MAVProxy/issues/11

I have tried the solution presented there, but to no effect. Is there a specific place to write the command param set ARMIN_CHECK 0? I write it in the window right after SITL is initiated ...Am I missing something here?

PS: if you want me to open this up as a separate issue then do let me know. 

Thanks in advance, 
Best
Akshay

SalahuddinAsh, 2016-10-20:
Hi everyone, 
I'm trying to run the first dronekit example. I did as @fnoop said and the error that appeared when running dronekit-sitl has disappeared. 

I run 
`dronekit-sitl /home/salahuddin/ardupilot/ArduCopter/ArduCopter.elf`
It run successfully and waited for connection

In another terminal window I run 
`python hello.py --connect tcp:localhost:5760` 
but it gave me the same error

`Start simulator (SITL)
Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
Traceback (most recent call last):
  File "hello.py", line 3, in <module>
    sitl = dronekit_sitl.start_default()
  File "/home/salahuddin/.local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 341, in start_default
    sitl.launch(sitl_args, await_ready=True, restart=True)
  File "/home/salahuddin/.local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 271, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error
`
What should I do?
Thanks

peterbarker, 2016-10-20:
On Thu, 20 Oct 2016, SalahuddinAsh wrote:

> Hi everyone,
> I'm trying to run the first dronekit example. I did as @fnoop said and the
> error that appeared when running dronekit-sitl has disappeared.
>
> I run
> dronekit-sitl [...]

This error message means that it isn't attempting to connect to your 
listening binary.  You can see "start_default" in the stack trace :-)

Here's an example that does take a connection string from the command 
line, and if it isn't found fires up dronekit-sitl:

https://github.com/dronekit/dronekit-python/blob/master/examples/follow_me/follow_me.py#L35

I'm thinking your hello.py doesn't have that argument, or it isn't being 
interpreted correctly.

SalahuddinAsh, 2016-10-20:
@peterbarker Thanks a lot for your fast reply. 
I'm trying to get started with DroneKit and I just copied and pasted the "Hello Drone" example [here](http://python.dronekit.io/guide/quick_start.html#basic-hello-drone). Could you please tell me how can I get this example to work, just to start with simple example.
Many thanks

peterbarker, 2016-10-20:
On Thu, 20 Oct 2016, SalahuddinAsh wrote:

> @peterbarker Thanks a lot for your fast reply.
> I'm trying to get started with DroneKit and I just copied and pasted the "Hello Drone" example here. Could you please tell me how can I get this example to work, just to start with simple example.

I could work with you to get that example working, but I think it would be 
better for you to run one of the other examples shipped with 
dronekit-python!

Can you locate "vehicle_state.py" in your installation?

Here's how I might do it:

pbarker@bluebottle:~$ locate vehicle_state.py
/home/pbarker/rc/dronekit-python/examples/vehicle_state/vehicle_state.py
pbarker@bluebottle:~$

If you're working on windows, you might try "Find Files or Folders".

When you find it, run it in the same way you attempted to run 
hello-drone.py.

SalahuddinAsh, 2016-10-20:
I'm sorry, but I couldn't find "vehicle_state.py". I run it in terminal but it gave nothing back!!
Could you give me link to it. I looked for it in Github but couldn't find it too.

peterbarker, 2016-10-20:
On Thu, 20 Oct 2016, SalahuddinAsh wrote:

> I'm sorry, but I couldn't find "vehicle_state.py". I run it in terminal but it gave nothing back!!
> Could you give me link to it. I looked for it in Github but couldn't find it too.

https://github.com/dronekit/dronekit-python/blob/master/examples/vehicle_state/vehicle_state.py

SalahuddinAsh, 2016-10-20:
Thanks a lot. It turned out that I didn't cloned the examples to my PC. It's working now. I'll try to understand the script. 
BTW, when I have question over DroneKit, [here](https://gitter.im/dronekit/dronekit-python) is the best place to ask, right?
Many Thanks

peterbarker, 2016-10-20:
On Thu, 20 Oct 2016, SalahuddinAsh wrote:

> 
> Thanks a lot. It turned out that I didn't cloned the examples to my PC. It's working now. I'll try to understand the script.
> BTW, when I have question over DroneKit, here is the best place to ask, right?

That's probably the best place, yes.

> Many Thanks

No worries :-)

weiminshen99, 2017-10-12:
Hi, All,

Thanks for all the posts. I need some help for running a native sitl copter on ODROID C0.
I built a native Copter on ubuntu 16.4 and run the following, but got a message that I have to specify a vehicle model. I tried with --model=quad, and still have this problem. Please help and really appreciate it.

$ dronekit-sitl /home/odroid/GitHub/ardupilot/ArduCopter/ArduCopte
os: linux, local binary: /home/odroid/GitHub/ardupilot/ArduCopter/ArduCopter.elf
Execute: /home/odroid/GitHub/ardupilot/ArduCopter/ArduCopter.elf --home=-35.363261,149.165230,584,353 --model=quad
Popen( ['/home/odroid/GitHub/ardupilot/ArduCopter/ArduCopter.elf', '--home=-35.363261,149.165230,584,353', '--model=quad']  cwd= /tmp/tmpJJ_D1w
You must specify a vehicle model
$

prash319, 2018-01-29:
I have the same problem and I am unable to solve it, 

I am trying to implement the basic "hello drone" script.

>>> print ( "Start simulator (SITL)" )
Start simulator (SITL)
>>> import dronekit_sitl.start_default()
SyntaxError: invalid syntax
>>> sitl = dronekit_sitl.start_default()
Traceback (most recent call last):
  File "<pyshell#2>", line 1, in <module>
    sitl = dronekit_sitl.start_default()
NameError: name 'dronekit_sitl' is not defined
>>> import dronekit_sitl
>>> sitl = dronekit_sitl.start_default()
Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
T_raceback (most recent call last):
  File "<pyshell#4>", line 1, in <module>
    sitl = dronekit_sitl.start_default()
  File "/usr/local/lib/python3.4/dist-packages/dronekit_sitl/__init__.py", line 341, in start_default
    sitl.launch(sitl_args, await_ready=True, restart=True)
  File "/usr/local/lib/python3.4/dist-packages/dronekit_sitl/__init__.py", line 271, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/lib/python3.4/subprocess.py", line 859, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.4/subprocess.py", line 1457, in _execute_child
    raise child_exception_type(errno_num, err_msg)
OSError: [Errno 8] Exec format error

Why am I getting this? How do I solve this?

I tried solving it by using the solution given by fnoop but when i do that my system is restarting.

hamzawinix, 2019-08-24:
this solved it for me
` sudo -H pip install dronekit-sitl
 `

nelsonifechukwu, 2023-05-26:
> It should be cd ardupilot/  and not cd ardupilot/ArduCopter