dronekit/dronekit-sitl #85: Simulating multiple drones with dronekit-sitl

lwalid, 2016-04-27:
Hi everyone I have a question,  can i simulate multiple drones with dronekit-sitl?
if yes, how we can do that?
for me i use ubuntu like OS.

peterbarker, 2016-04-27:
On Wed, 27 Apr 2016, lwalid wrote:

> Hi everyone I have a question, can i simulate multiple drones with
> dronekit-sitl?

https://github.com/peterbarker/dronekit-python/tree/source-system-filtering/examples/multivehicle

> if yes, how we can do that?
> for me i use ubuntu like OS.

This work is still out-of-tree.  I have work to do in pymavlink to get 
things acceptable upstream.

developermastermind07, 2019-04-24:
use -I0,I1,... options in dronekit-sitl 
eg
$ dronekit-sitl -I0 
$ dronekit-sitl -I1
...