dronekit/dronekit-sitl #86: Flightgear integration not present

randomProgrammer, 2016-05-01:
I'm successfully running dronekit-sitl on OS X El Capitan, but as far as I can tell the precompiled binaries do not support sending data to flightgear which is handy for visualisation. This seems to be supported by the legacy SITL, see fgout arg: 

`# pysim args
print('Note: Starting pysim for legacy SITL.')
simargs = [sys.executable, os.path.join(os.path.dirname(__file__), 'pysim/sim_wrapper.py'),
'--simin=127.0.0.1:5502', '--simout=127.0.0.1:5501', '--fgout=127.0.0.1:5503',`

Is this meant to be supported? Is there any way to compile the binaries to allow this argument?

hamishwillee, 2016-05-01:
Yes, not supported for the current binaries. The [Ardupilot documentation](http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html#flightgear-3d-view-optional) states:

> FlightGear support is currently only in master (January 2016). It should appear in the next versions of the vehicle codelines (not present on current versions: Copter 3.3, Plane 3.4, Rover 2.5).

hamishwillee, 2016-05-01:
It would be good to get a rebuild to latest platforms @mrpollo ?

jainakshay91, 2016-05-02:
@hamishwillee so at the moment there is no method to visualize the simulations with dronekit-sitl?

hamishwillee, 2016-05-02:
@jainakshay91 If you only want 2d then you can "visualise" the path in your GCS. If you do want 3D as per flightgear then you can't use DK sitl existing builds. If you wanted these with DK-SITL you'd have to build your own binaries (this is perfectly possible) but if visualisation is your main point you might as well just do a native build without DK-SITL.