dronekit/dronekit-sitl #89: relaunch-preserve-eeprom branch doesn't seem to work

sciencectn, 2016-05-18:
I added a test to my fork which tries to change SYSID_THISMAV and then verifies that it actually changed:
https://github.com/sciencectn/dronekit-sitl/blob/relaunch-preserve-eeprom/tests/test_sitl.py

Maybe I'm just doing it wrong, but the test fails and SYSID_THISMAV just remains at 1.0 when you re-open SITL. 

This came about because I was trying to get peterbarker's multivehicle example to work:
https://github.com/peterbarker/dronekit-python/tree/source-system-filtering/examples/multivehicle

peterbarker, 2016-05-18:
On Wed, 18 May 2016, Chris Taylor wrote:

> I added a test to my fork which tries to change SYSID_THISMAV and then verifies that it actually changed: [...]

I'm currently heavily using that example for development work I am 
currently doing.  I will push up all of the associated branches later 
today so you're using what I'm using, which should make fixing this for 
you easier!

peterbarker, 2016-05-19:
On Wed, 18 May 2016, Chris Taylor wrote:

> I added a test to my fork which tries to change SYSID_THISMAV and then verifies that it actually changed:
> https://github.com/sciencectn/dronekit-sitl/blob/relaunch-preserve-eeprom/tests/test_sitl.py

Many thanks for that!  I've taken the liberty of cherry-picking that test 
into my multi-vehicle-wip branch - I hope you're OK with that?

> Maybe I'm just doing it wrong, but the test fails and SYSID_THISMAV just remains at 1.0 when you re-open SITL.

There were a few issues:
  - you were creating a new SITL object rather than reusing the existing 
one
  - there was no delay after changing the parameter to let the ArduPilot 
instance get the new parameter data to disk
  - in the second launch() call you didn't specify which vehicle to connect 
to.  It would wait for vehicle 1 to pop up, which was never going to 
happen since you'd given it a new ID of 10!
  - You didn't specify the working directory for the launch as the old 
working directory for the SITL instance.  This was a wart in the API which 
I've fixed in the multivehicle branch now, so that is now not necessary.

My multivehicle-wip branch is here:
https://github.com/dronekit/dronekit-sitl/compare/peterbarker-multivehicle-wip?expand=1

sciencectn, 2016-05-19:
Awesome, thanks for your additions. 

I'm confused on this though:
> you were creating a new SITL object rather than reusing the existing one

Don't you need to relaunch it so it reads the values back from the EEPROM file? If I just change SYSID_THISMAV it doesn't seem to take effect until the next launch of SITL, at least from testing with the regular sim_vehicle.sh in ArduPilot. 

>  in the second launch() call you didn't specify which vehicle to connect
> to.  It would wait for vehicle 1 to pop up, which was never going to
> happen since you'd given it a new ID of 10!

So I guess the reason it worked it because SYSID_THISMAV never went into effect. 

I also checked it by just looking the raw mavlink packet header with netcat, System ID and Component ID are definitely stuck at 1.

peterbarker, 2016-05-19:
On Wed, 18 May 2016, Chris Taylor wrote:

> I'm confused on this though:
>       you were creating a new SITL object rather than reusing the existing one
>
> Don't you need to relaunch it so it reads the values back from the EEPROM file? If I just [...]

Yep, you definitely need to relaunch it.  OTOH, you need to relaunch *it*, 
not launch another dronekit-sitl instance! :-)

This is the offending line I removed:
https://github.com/dronekit/dronekit-sitl/commit/60f5843ee0ad5c5cbb7bb55f7385c35d21f076a7#diff-ff4cf282ccba391cd1f810efaa3a86bdL69

> So I guess the reason it worked it because SYSID_THISMAV never went into effect.
> 
> I also checked it by just looking the raw mavlink packet header with netcat, System ID and Component ID are definitely stuck at 1.

Yep, I would believe that.  You were launching an entirely new SITL 
instance :-)

You may have expected "use_saved_data" to persist across SITL instances? 
It doesn't, at least not without throwing more parameters at it. 
use_saved_data persists data across calls to "launch", not across SITL 
instances.

sciencectn, 2016-05-19:
Sweet, it works now. I gave it a working directory and reused the SITL instance. 

I have a new issue now: it won't arm and I get ">>> PreArm: RC not calibrated" if I use use_saved_data=True. I'll post this in a separate issue with a test.