dronekit/dronekit-sitl #90: Cannot arm with use_saved_data=True

sciencectn, 2016-05-19:
I just get a ton of ">>> PreArm: RC not calibrated" errors if I use this option. It works fine otherwise. 

Here's my test:
https://github.com/sciencectn/dronekit-sitl/blob/relaunch-preserve-eeprom/tests/test_sitl.py

peterbarker, 2016-05-20:
On Thu, 19 May 2016, Chris Taylor wrote:

> I just get a ton of ">>> PreArm: RC not calibrated" errors if I use this option. It works fine otherwise.
> 
> Here's my test:
> https://github.com/sciencectn/dronekit-sitl/blob/relaunch-preserve-eeprom/tests/test_sitl.py

You're attempting to use "use_saved_data" on the first call to launch(). 
At that point there's no "saved data" to use.  This means the eeprom will 
be empty, and you don't have any calibrations in there (if 
"use_saved_data" is not present then we copy a default eeprom into place 
(or, in my multivehicle-wip branch, use a "defaults file", which allows 
you to work with an ArduPilot binary you've compiled yourself)).

We should probably throw an error if you attempt to "use saved data" 
and there's no actual saved data in the working directory.

sciencectn, 2016-05-20:
Ah that makes sense. Yeah an error would be nice for us noobs.