dronekit/dronekit-sitl #92: Download DroneKit-SITL error

Mario-Luis, 2016-06-03:
My computer is on window 8.1. After I setup python2.7.11(it will setup pip automatically), I type "pip install dronekit-sitl -UI" [here](http://python.dronekit.io/develop/sitl_setup.html) in my cmd.exe, it will turn out some error as follow:
```
Command "d:\python27\python.exe -c "import setuptools, tokenize;__file__='c:\\us
ers\\mario\\appdata\\local\\temp\\pip-build-ddozcc\\psutil\\setup.py';exec(compi
le(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __fil
e__, 'exec'))" install --record c:\users\mario\appdata\local\temp\pip-dd_aeg-rec
ord\install-record.txt --single-version-externally-managed --compile" failed wit
h error code 1 in c:\users\mario\appdata\local\temp\pip-build-ddozcc\psutil
```
But I don't what had happened. Did I do something wrong? I hope someone can help me.
Thank you for reading.

lroberts99, 2016-06-03:
I have a similar issue when downloading the SITL. I am on OS X El Capitan.

`Command "/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u -c "import setuptools, tokenize;__file__='/private/var/folders/b3/1jhbz5092679pgb_kdccn7dr0000gn/T/pip-build-k5Ug8L/pymavlink/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/b3/1jhbz5092679pgb_kdccn7dr0000gn/T/pip-s83lhJ-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/b3/1jhbz5092679pgb_kdccn7dr0000gn/T/pip-build-k5Ug8L/pymavlink/`

Any help would be great.

hamishwillee, 2016-06-05:
@peterbarker Any thoughts on this one?

peterbarker, 2016-06-06:
Not enough information in that command line to determine the problem - "error code 1" is not particularly informative.

Perhaps an archive of the build directory will contain more information - "/private/var/folders/b3/1jhbz5092679pgb_kdccn7dr0000gn/T/pip-build-k5Ug8L/pymavlink/" in the second example, c:\users\mario\appdata\local\temp\pip-build-ddozcc\psutil in the first.

I did notice a random pip package failing to build on the ArduPilot Vagrant VM the other day - it may be related, and magically fixed by now.