dronekit/dronekit-sitl #93: OSError: [Errno 8] Exec format error

alduxvm, 2016-06-06:
Hi!

I just did a fresh install of Ubuntu LTS 14.04 on a virtual machine, and installed using `sudo pip install dronekit-sitl`. 

When trying to open a instance, I get this error:
```
aldux@ubuntu:~/Flight-Stack/simulator$ dronekit-sitl copter --home=55.870595,-4.287639,0,0
Using MAVLink 1.0
os: linux, apm: copter, release: stable
Downloading SITL from http://dronekit-assets.s3.amazonaws.com/sitl/copter/sitl-linux-copter-3.3.tar.gz
Extracted.
Execute: /home/aldux/.dronekit/sitl/copter-3.3/apm --home=55.870595,-4.287639,0,0 --model=quad
Traceback (most recent call last):
  File "/usr/local/bin/dronekit-sitl", line 9, in <module>
    load_entry_point('dronekit-sitl==3.0.1', 'console_scripts', 'dronekit-sitl')()
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 417, in main
    sitl.launch(args, verbose=True)
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 239, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error
```

Appears that dronekit-sitl is identifying my system as 'win32'... 

 Cheers!

hamishwillee, 2016-06-06:
One potential issue is that I think the version of PyPi is out of date. Try install sitl  from github:
```
pip install git+https://github.com/dronekit/dronekit-sitl
```

alduxvm, 2016-06-06:
Ok, I uninstalled `dronekit-sitl`, then I did: `sudo pip install git+https://github.com/dronekit/dronekit-sitl`

Then again tried to open it:
```
aldux@ubuntu:~$ dronekit-sitl copter --home=55.870595,-4.287639,0,0
os: linux, apm: copter, release: stable
Downloading SITL from http://dronekit-assets.s3.amazonaws.com/sitl/copter/sitl-linux-copter-3.3.tar.gz
Download Complete.
Payload Extracted.
Ready to boot.
Execute: /home/aldux/.dronekit/sitl/copter-3.3/apm --home=55.870595,-4.287639,0,0 --model=quad
Traceback (most recent call last):
  File "/usr/local/bin/dronekit-sitl", line 9, in <module>
    load_entry_point('dronekit-sitl==3.0.1', 'console_scripts', 'dronekit-sitl')()
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 475, in main
    sitl.launch(args, verbose=True)
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 266, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error
```

same stuff... :(

hamishwillee, 2016-06-06:
Its a bugger. I've not been able to reproduce this on this same ubuntu in a VM :-( 

@peterbarker Any suggestions? We've seen this before, but I personally can't get it to "fail".

alduxvm, 2016-06-06:
I had an old VM with Ubuntu 14.04... updated and the installed and worked excellent...

Then I created a new one, brand new, "fresh" and this problem started...

hamishwillee, 2016-06-06:
I just completely reinstalled VM on Windows 10 of Ubuntu 14.04 and it worked fine ..... As you say, same Ubuntu 14.04. I don't doubt that you have a problem, I just can't get it to fail in the same way.

alduxvm, 2016-06-07:
I manage to get a work-around for this situation... I compiled ArduPilot like this:

```
git clone git://github.com/ArduPilot/ardupilot.git
cd ardupilot/ArduCopter
make -j4 sitl
```

And then to execute: `dronekit-sitl ./ardupilot/ArduCopter/ArduCopter.elf `

Is not ideal... or as easy as just using pip, but is working...

hamishwillee, 2016-06-07:
Interesting. The fact that you can launch native binaries implies that the original error report is accurate - i.e. that you've been downloading binaries that aren't native or aren't detected as native. @peterbarker Any ideas how we might debug this?

peterbarker, 2016-06-07:
On Tue, 7 Jun 2016, Hamish Willee wrote:

> Interesting. The fact that you can launch native binaries implies that the original error report is accurate - i.e. that you've been downloading binaries
> that aren't native or aren't detected as native. @peterbarker Any ideas how we might debug this?

We should add a check into dronekit-sitl which warns when you try to 
download on a non-supported platform and dies with a clear error when you 
try torun on  non-supported platform.

But first we should do a fresh release of dronekit-sitl which includes the 
start_default entry point.  I haven't worked out why people are getting a 
dronekit-python from the git repository when using pip with default 
settings.

hamishwillee, 2016-06-07:
OK, can you make that release? ... or do you not have the magic power?

ggregory8, 2016-08-22:
Hi, I'm having the same issue on Ubuntu 14.04.4 LTS using master. Any news on this?

jarpii, 2016-12-27:
Traceback (most recent call last):
  File "../../base/android/jni_generator/jni_generator.py", line 1109, in <module>
    sys.exit(main(sys.argv))
  File "../../base/android/jni_generator/jni_generator.py", line 1105, in main
    GenerateJNIHeader(input_file, output_file, options)
  File "../../base/android/jni_generator/jni_generator.py", line 1022, in GenerateJNIHeader
    input_file, options)
  File "../../base/android/jni_generator/jni_generator.py", line 576, in CreateFromFile
    return JNIFromJavaSource(contents, fully_qualified_class, options)
  File "../../base/android/jni_generator/jni_generator.py", line 531, in __init__
    contents = self._RemoveComments(contents)
  File "../../base/android/jni_generator/jni_generator.py", line 564, in _RemoveComments
    stderr=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error

hamishwillee, 2016-12-28:
Almost certainly running SITL on an ARM platform. Binaries are no supplied, though you can build your own.

jarpii, 2016-12-28:
What could I do?
It never happened to me.

kevinjhur, 2017-01-20:
I think this error is from that dronekit sitl can not support 32 bit linux OS.
Below is my try, please refer.
on Ubuntu 14.04 64bit with intel PC -> works fine
on Ubuntu mate 16.04 with Raspberry -> works fine
on Ubuntu 14.04 with Odroid ux4 -> Error same as above

Ahmad-Drak, 2017-01-29:
Same issue here. Tried it on Odroid XU4 running Mate 16.04 resulted in the same error. Any fix for that?

PavaniLathaB, 2017-10-17:
Hi,
Can any help me to solve the below issue : 
Traceback (most recent call last):
  File "../../base/android/jni_generator/jni_generator.py", line 1109, in <module>
    sys.exit(main(sys.argv))
  File "../../base/android/jni_generator/jni_generator.py", line 1105, in main
    GenerateJNIHeader(input_file, output_file, options)
  File "../../base/android/jni_generator/jni_generator.py", line 1018, in GenerateJNIHeader
    jni_from_javap = JNIFromJavaP.CreateFromClass(input_file, options)
  File "../../base/android/jni_generator/jni_generator.py", line 521, in CreateFromClass
    stderr=subprocess.PIPE)
  File "/usr/local/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/local/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
external/chromium_org/ui/gl/surface_jni_headers.target.linux-arm.mk:22: recipe for target '/home/vipul/files/out/target/product/bacon/obj/GYP/shared_intermediates/ui/gl/jni/Surface_jni.h' failed
make: *** [/home/vipul/files/out/target/product/bacon/obj/GYP/shared_intermediates/ui/gl/jni/Surface_jni.h] Error 1
make: *** Waiting for unfinished jobs....
external/chromium_org/third_party/sqlite/amalgamation/sqlite3.c: In function 'fts3SnippetFunc':
external/chromium_org/third_party/sqlite/amalgamation/sqlite3.c:124201:11: warning: 'iS' may be used uninitialized in this function [-Wmaybe-uninitialized]
external/chromium_org/third_party/sqlite/amalgamation/sqlite3.c:124193:13: note: 'iS' was declared here
external/chromium_org/third_party/sqlite/amalgamation/sqlite3.c: In function 'recoverOpen':
external/chromium_org/third_party/sqlite/amalgamation/sqlite3.c:112080:24: warning: 'pLeafCursor' may be used uninitialized in this function [-Wmaybe-uninitialized]
external/chromium_org/third_party/sqlite/amalgamation/sqlite3.c: In function 'unixOpen':
external/chromium_org/third_party/sqlite/amalgamation/sqlite3.c:29202:21: warning: 'openMode' may be used uninitialized in this function [-Wmaybe-uninitialized]
external/chromium_org/third_party/sqlite/amalgamation/sqlite3.c:29195:12: note: 'openMode' was declared here

Thanks in advance :)

sameenashaikh29, 2017-12-15:
I am trying to interface the raspberry pi with the pixhawk through serial pins on raspberry pi. I am able to get a connection between the two. 

I get the following messages after setting up the connection
i@raspberrypi:~ $ sudo -s
root@raspberrypi:/home/pi# mavproxy.py --master=/dev/ttyACM0 --baudrate 57600 --aircraft plane
Connect /dev/ttyACM0 source_system=255
no script plane/mavinit.scr
Log Directory: plane/logs/2017-12-15/flight1
Telemetry log: plane/logs/2017-12-15/flight1/flight.tlog
MAV> Waiting for heartbeat from /dev/ttyACM0
*o]l`:H9%online system 1
ALT_HOLD> Mode ALT_HOLD
[S$<f1z9;>;APM: APM:Copter V3.5.4 (284349c3)
APM: PX4: 0384802e NuttX: 1bcae90b
APM: Frame: QUAD
APM: PX4v3 00490021 33355119 38393235
Received 773 parameters
Saved 773 parameters to plane/logs/2017-12-15/flight1/mav.parm
fence breach
mode
ALT_HOLD> ('Available modes: ', ['RTL', 'POSHOLD', 'LAND', 'OF_LOITER', 'STABILIZE', 'AUTO', 'GUIDED', 'THROW', 'DRIFT', 'FLIP', 'AUTOTUNE', 'ALT_HOLD', 'BRAKE', 'LOITER', 'AVOID_ADSB', 'POSITION', 'CIRCLE', 'SPORT', 'ACRO'])
 
After this i tried writing a python code to get some data from pixhawk
I followed the steps here
http://python.dronekit.io/guide/quick_start.html#installation

but then i get this error
pi@raspberrypi:~ $ sudo python hello.py
Start simulator (SITL)
Starting copter simulator (SITL)
Downloading SITL from http://dronekit-assets.s3.amazonaws.com/sitl/copter/sitl-linux-copter-3.3.tar.gz
Download Complete.
Payload Extracted.
Ready to boot.
Failed to run (/root/.dronekit/sitl/copter-3.3/apm)
Traceback (most recent call last):
  File "hello.py", line 3, in <module>
    sitl = dronekit_sitl.start_default()
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 454, in start_default
    sitl.launch(sitl_args, await_ready=True, restart=True)
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 253, in launch
    caps = ArdupilotCapabilities(self.path)
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 165, in __init__
    self.has_defaults_path = "--defaults path" in helptext
UnboundLocalError: local variable 'helptext' referenced before assignment

Any help would be appreciated. Stuck at this from quiet long.

Thanks & Regards

sameenashaikh29, 2017-12-15:
@hamishwillee :+1:
In issue #95 : you said we cant use it on an arm processor. So using raspberry pi as ssh and using on my laptop with intel processor would have a difference???

hamishwillee, 2017-12-18:
Yes. Only intel binaries are supplied.

Slight clarification though. You can't use ARM "out of the box" because no binaries are built for ARM processors. If you wanted though, you could build the binaries yourself on the platform you want to run them, and then use SITL to launch them.

mdb2301, 2021-11-07:
Everyone's saying "You can build your own binaries for ARM platform". Please explain HOW???

vitorstone, 2021-11-18:
I'm getting the same thing on my Raspberry Pi 

Traceback (most recent call last):
  File "test.py", line 2, in <module>
    sitl = dronekit_sitl.start_default()
  File "/home/pi/.local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 462, in start_default
    sitl.launch(sitl_args, await_ready=True, restart=True)
  File "/home/pi/.local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 251, in launch
    caps = ArdupilotCapabilities(self.path)
  File "/home/pi/.local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 160, in __init__
    process = subprocess.Popen([path, '--help'], stdout=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error