dronekit/dronekit-sitl #95: how to use dronekit sitl  on raspberry pi

keyurrakholiya, 2016-06-09:
can i use dronekit sitl on raspberry pi with rasbian? or should i use ubuntu on raspberry pi? dronekit sitl is supported on R-pi or not?

thanks!

LeonidasAntoniou, 2016-08-07:
The RPi has an ARM-based system architecture. The requirement for SITL to run in Linux is to have an Intel processor (x86 architecture) ([here](http://python.dronekit.io/develop/sitl_setup.html))

So, to cover your question, you cannot use dronekit sitl on RPi with raspbian or ubuntu or any linux distribution, since RPi does not have the proper processor (Intel x86).

What if you install Windows 10 on RPi and then trying to run sitl again? I have not tried it but it could be a possible solution if you want to run it so badly.

hamishwillee, 2016-08-07:
To be more precise, DK-SITL does not **provide** binaries for ARM based architectures. You can build your own SITL binaries on RPi you can use DK-SITL to launch them. 

The main use case for DK-SITL is to run SITL on a host computer during development so that you can test your DK-Python code without running it on a real device. That development computer is far more likely to be an x86 desktop or laptop than an RPi/x86 based machine.

closed 2016-08-07 by hamishwillee

fraseg, 2019-07-22:
Is there anywhere where I can get the build instructions to make a version for the pi?

djdylex, 2020-10-22:
> Is there anywhere where I can get the build instructions to make a version for the pi?

As for building it, just need to compile the files using make. Does anyone know what to do after that? Like where do you put the binaries?

Kanishk598, 2022-04-02:
@djdylex Were you able to do this properly?

HefnySco, 2022-04-03:
you can compile ardupilot-sitl on arm easily using cross compile.
or just download ardupilot on RPI and compile it "make sitl" and you have an SITL running on RPI.