dronekit/dronekit-sitl #100: Setting up a Simulated Vehicle (SITL)

jhljx, 2016-08-15:
when I type the instruction 'dronekit-sitl copter' in the terminal, it show that:
Traceback (most recent call last):
  File "/usr/local/bin/dronekit-sitl", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 475, in main
    sitl.launch(args, verbose=True)
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 266, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error

How to fix this error??

hamishwillee, 2016-08-15:
This basically says that the system things the binaries you're downloading are not appropriate for your platform. The usual cause of this error is trying to run SITL on an ARM platform rather than an x86 platform. 

If you've already managed to build SITL then you can point dronekit-sitl at those binaries and launch them.

ikhwanrobootic, 2016-09-11:
Hiii @hamishwillee. I have problem with install sitl in my linux ubuntu. I work with linux ubuntu 14.04. I'm follow step by step instruction in site. But i have some problem with sim_vehicle.py: command not found. Why ???. Please answer questiin form me. Thx and sorry with my language.

hamishwillee, 2016-09-11:
@ikhwanrobootic if you're using dronekit-sitl then you should not even be using sim_vehicle.py - what is the page you are following?

ikhwanrobootic, 2016-09-12:
@hamishwille i 'm follow step by step in http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html 
I've followed the steps begin "download ardupilot", "install jsbsim", "install some required packeges", and "add some directories to your search path". All step it's ok. But, and when i type "sim_vehicle.py -w" in terminal out message error "sim_vehicle.py : command not found"

hamishwillee, 2016-09-12:
@ikhwanrobootic OK, hence the first confusion - this is the area for dronekit-sitl, and you're posting about generic SITL. The best place to get support for this problem is the discussion boards.

That said, this error means that the file can't be found. This probably means that you didn't following the instructions in this section: http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html#add-some-directories-to-your-search-path (the file is in $HOME/ardupilot/Tools/autotest , and you need to add that to your .bashrc path)

Regards

ikhwanrobootic, 2016-09-12:
Oke @hamishwille. Can you tell me how to add .bashrc path. In my pc, name for dir is $home not $HOME. If this is cause sim_vehicle.py not working in my linux. 
Even though. I have include my improve to path as "export PATH=$PATH:$home/ardupilot/Tools/autotest" 
But this step still not working for me.
Can i know your email. So i could teel you about my problem and my screen shoot in my linux. 
Thx soo much.

hamishwillee, 2016-09-12:
Hi @ikhwanrobootic , nope, I don't share my email for technical support. As per my last email, this is not part of this area, and the right place to ask in the discussion boards. That exposes you to a lot more people who might be able to help you .

Before I close this issue, you may have added your path, but did you reload the .bashrc (e.g. follow the instruction to call `. ~/.bashrc`). If not then the command won't yet be in your bash prompt, even though you added the export.

closed 2016-09-12 by hamishwillee

ikhwanrobootic, 2016-09-12:
Ok thank you so much for your help. @hamishille

hamishwillee, 2016-09-12:
You're most welcome. Hope that helps ... it is about the extent of my knowledge :-)

escker, 2017-04-24:
@hamishwillee 
**My goal:** To set home location manually after arming and taking off as well to trigger the RTL mode

import time
from dronekit import connect, VehicleMode, LocationGlobalRelative
import modules.UDPserver as udp
from modules.utils import *
from modules.pixVehicle import *
vehicle = connect('/dev/ttyAMA0', wait_ready=True)
arm_and_takeoff(vehicle, 10)
vehicle.home_location(12.732824,80.073488, 0)
vehicle.mode = VehicleMode("RTL")
time.sleep(10)

**But the vehicle.home_location line doesn't seems to work... Can u please help me with a solution?**

hamishwillee, 2017-04-24:
That line can fail silently if the home location is more than 50km than your current position. I added a PR that has never been accepted that checks for stuff like that.