dronekit/dronekit-sitl #101: Additional build server: firmware.diydrones.com

djnugent, 2016-08-22:
The default host server does not include beta builds. It would be nice to access the build server at firmware.diydrones.com, instead of cloning the ardupilot repo, building it manually, and running a custom binary.
 
Is this something other people desire?

hamishwillee, 2016-08-23:
If we can get reliable SITL builds, I don't care where they come from, and yes, it would be great to have a full and reliable set for testing. So yes, this would be awesome.

BUT, does firmware.diydrones.com create native **SITL** builds, and for all platforms?

djnugent, 2016-08-23:
I don't believe it does now that you mention it. 

Does the current firmware server update automatically? Because the images are out of date(at least for rover). Maybe the real solution is to increase the variety of binaries hosted on the current server.

hamishwillee, 2016-08-23:
Absolutely we need more binaries, and some testing of them. There is no automated build process and there should be (I *think* there is an issue for this). I'd actually very much like SITL builds to be done as part of the "official builds".

peterbarker, 2016-08-23:
The firmware server doesn't keep around SITL builds - and only creates them for the host platform during testing.

Note that creating a dronekit-sitl-suitable build involves extra steps (https://github.com/dronekit/dronekit-sitl/tree/master/stage).  I've never done that.  Having additional options to grab sitl binaries and parameters from firmware.ardupilot.org is probably a good way forward there.