dronekit/dronekit-sitl #105: add latest plane/copter versions

expntly, 2016-09-18:
As of Sep 18 2016

> $ dronekit-sitl --list
> solo-1.2.0
> solo-2.0.18
> solo-2.0.20
> plane-3.3.0
> copter-3.3

APM Plane and Copter are respectively at version 3.7.0 and  3.3.3.  Is there a way to (manually?) use these versions?

@peterbarker  knows maybe?

hamishwillee, 2016-09-18:
We don't supply the binaries for those versions yet. @mrpollo is looking at automating binary creation.

>APM Plane and Copter are respectively at version 3.7.0 and 3.3.3. Is there a way to (manually?) use these versions?

Yes, you can manually build the binaries from source using the [normal methods for your platform](http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html) and then launch them using dk-sitl as below.
```
dronekit-sitl /path/to/local/binary [parameters]
```
@mrpollo - can you please comment on your progress to building new binaries?

okalachev, 2016-12-01:
@hamishwillee, I wanted to build a binary for macOS by myself, but unfortunately I haven't found any tutorial for this on your link. 
Can you please help me with finding such a tutorial?

hamishwillee, 2016-12-01:
@okalachev I have never done this. The only instructions I know of are: https://github.com/dronekit/dronekit-sitl/tree/master/stage#building-a-new-sitl-image

okalachev, 2016-12-02:
@hamishwillee Thanks.
But anyway I couldn't have built this with some C++ compilation error :(

Javiercerna, 2017-12-24:
Hi, I managed to compile a SITL image for the latest stable Arducopter version (V3.5.4). I would like to help, but I'm not sure if this build will only work on my computer (Ubuntu Bash on Windows), or if I can safely upload it here.

Here's a screenshot of how I'm calling it (the regular way, using dronekit-sitl and the path to the compiled image).

[image]

peterbarker, 2018-01-03:
On Sun, 24 Dec 2017, Javier Cerna wrote:

> Here's a screenshot of how I'm calling it (the regular way, using
> dronekit-sitl and the path to the compiled image).

Thanks!  I'm wondering how successful we might be in adding SITL binaries 
to ArduPilot's autotest server....

Javiercerna, 2018-02-04:
Is there anything I can do to help? I regularly use this library, and I would like to contribute regarding the latest copter versions.

zarar7576, 2019-08-19:
I made this yesterday, these are binaries and EEPROM for the latest copter firmware as of now at least
[firmware_36.zip](https://github.com/dronekit/dronekit-sitl/files/3514761/firmware_36.zip)