dronekit/dronekit-sitl #106: QGroundControl + Dronekit-sitl + mavproxy: ERROR: Floating point exception - aborting

alduxvm, 2016-09-22:
Hi! 

Is the support with qgroundcontrol limited? I'm trying to upload a mission to a SITL vehicle, but when its being sent, dronekit-sitl crashes. Allow me to explain my setup. I'm on mac, using the newest dronekit-sitl, mavproxy and qgroundcontrol 3.0.1.

I start my SITL instance with:
`dronekit-sitl copter`

Then I start mavproxy like this:
`mavproxy.py --master=tcp:127.0.0.1:5760 --out=udpout:127.0.0.1:14550`

Then I open qgroundcontrol and connect to that udp port... Then I try to upload to the vehicle a mission created in qgroundcontrol, and it gives me this error:

[image]

And then dronekit-sitl crashes and sends this output:
```
dronekit-sitl copter --home=55.870595,-4.287639,0,0
os: osx, apm: copter, release: stable
SITL already Downloaded and Extracted.
Ready to boot.
Execute: /Users/2049632v/.dronekit/sitl/copter-3.3/apm --home=55.870595,-4.287639,0,0 --model=quad
bind port 5760 for 0
Started model quad at 55.870595,-4.287639,0,0 at speed 1.0
Serial port 0 on TCP port 5760
Starting sketch 'ArduCopter'
Starting SITL input
Waiting for connection ....
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763

ERROR: Floating point exception - aborting
```

I'm doing something wrong, or simply there is not support for qgroundcontrol and sitl?

In qgroundcontrol I have the "Ardupilot firmware" setup for offline mission editing.

peterbarker, 2016-09-22:
On Thu, 22 Sep 2016, Aldo Vargas wrote:

> ERROR: Floating point exception - aborting

That is an error from the ardupilot binary.  This is very much Not Good.

> I'm doing something wrong, or simply there is not support for qgroundcontrol
> and sitl?
> 
> In qgroundcontrol I have the "Ardupilot firmware" setup for offline mission
> editing.

So what's probably happening is that a division-by-zero error is occuring 
due to some bodgy values in the mission.  Some protection has gone into 
ArduPilot master to protect against some of these, but it is possible you 
have found a new one!

Do you think you might be able to give us a stack trace, or at least a 
core file?  Since ArduPilot is abort(3)ing, if you have your coresize 
limit large enough (e.g. ulimit -c unlimited) you should get a core file.

Failing that, a copy of the mission.txt file would possibly work.

alduxvm, 2016-09-23:
Ok, first this are the versions I'm using:
```
> mavproxy.py --version
MAVProxy is a modular ground station using the mavlink protocol
MAVProxy Version: 1.5.4
```
```
> dronekit-sitl --version
3.1.0
```
QGroundcontrol: v3.0.1

Corezise:
```
> ulimit
unlimited
```

What's the procedure to get a stack trace? where should I put the code to print it?

This is a screenshot of the error:
[image]

And the mission file is here:
[test.txt](https://github.com/dronekit/dronekit-sitl/files/489225/test.txt)