dronekit/dronekit-sitl #109: Strange behavior with speed up.

ianamason, 2016-12-12:
Short description:

The following behavior is  entirely repeatable.

If I launch a drone:
```
 dronekit-sitl copter-3.3 --instance  0 --home=-7.162675,-34.817705,36,250
```
and tell it to takeoff, then move a small distance, it does so. On reaching its
destination it hovers in GUIDED mode (even after its battery level goes negative).

In contrast, when I launch the drone via
```
 dronekit-sitl copter-3.3 --instance  0 --home=-7.162675,-34.817705,36,250 --speedup=3
```
and issue the same commands the drone starts to move, but then very quickly goes into
RTL mode and goes home and land.

If you need more details let me know. I will happily supply them. We are trying to use dronekit-sitl
to do statistical analysis of a control system, and not being able to use `speedup` is very 
inconvenient.

peterbarker, 2016-12-13:
On Mon, 12 Dec 2016, Ian A Mason wrote:

> If you need more details let me know. I will happily supply them. We are
> trying to use dronekit-sitl
> to do statistical analysis of a control system, and not being able to use
> speedup is very
> inconvenient.

I believe this problem is present on normal SITL (sim-vehicle.py).

Try setting the SIM_SPEEDUP parameter instead of specifying --speedup on 
the commandline.

ianamason, 2016-12-13:
Thanks, I'll give that a whirl ...

ianamason, 2016-12-13:
Any hints on how to do this over a mavlink?  I have tried:

```
vehicle.message_factory.param_set_send(0, 0, 'SIM_SPEEDUP', 3, mavutil.mavlink.MAVLINK_TYPE_INT32_T)
```

peterbarker, 2016-12-13:
On Tue, 13 Dec 2016, Ian A Mason wrote:

> Any hints on how to do this over a mavlink? I have tried:

while vehicle.parameters['SIM_SPEEDUP'] != 3:
   print("Setting SIM_SPEEDUP")
   vehicle.parameters['SIM_SPEEDUP'] = 3
   time.sleep(0.1)

ianamason, 2016-12-13:
Sorry ahead of time for being thick, but that seems to give me a key error:
```
/usr/local/lib/python2.7/dist-packages/dronekit/__init__.pyc in __getitem__(self, name)
   2390         name = name.upper()
   2391         self.wait_ready()
-> 2392         return self._vehicle._params_map[name]
   2393
   2394     def __setitem__(self, name, value):

KeyError: 'SIM_SPEEDUP'

```

I tried modifying it a tad to 

```
while True:
    self.vehicle.parameters['SIM_SPEEDUP'] = value
    print("Setting SIM_SPEEDUP")
    if self.vehicle.parameters['SIM_SPEEDUP'] == value:
        break
    time.sleep(0.1)
```

but I still get the key error in the lookup in the if.  I looked at the dictionary and the
only `SIM_` keys I see are 

```
SIM_WIND_TURB
SIM_ARSP_FAIL
SIM_BATT_VOLTAGE
SIM_DRIFT_TIME
SIM_GPS_HZ
SIM_GPS_DELAY
SIM_MAG_ERROR
SIM_FLOW_RATE
SIM_GYR_RND
SIM_RC_FAIL
SIM_MAG_RND
SIM_WIND_DELAY
SIM_SONAR_RND
SIM_SONAR_SCALE
SIM_DRIFT_SPEED
SIM_WIND_SPD
SIM_GPS_GLITCH_Y
SIM_GPS_GLITCH_X
SIM_GPS_GLITCH_Z
SIM_MAG_DELAY
SIM_GPS_DISABLE
SIM_ACC2_RND
SIM_FLOW_ENABLE
SIM_BARO_DISABLE
SIM_SERVO_RATE
SIM_SONAR_GLITCH
SIM_ENGINE_MUL
SIM_GPS_TYPE
SIM_ACC_RND
SIM_ACC_BIAS_Z
SIM_BARO_DRIFT
SIM_ACC_BIAS_X
SIM_ACC_BIAS_Y
SIM_TERRAIN
SIM_BARO_DELAY
SIM_GPS_DRIFTALT
SIM_BARO_GLITCH
SIM_BARO_RND
SIM_FLOAT_EXCEPT
SIM_MAG_OFS_Z
SIM_MAG_OFS_X
SIM_MAG_OFS_Y
SIM_ACCEL_FAIL
SIM_GPS2_ENABLE
SIM_GPS_NUMSATS
SIM_MAG_MOT_X
SIM_MAG_MOT_Y
SIM_MAG_MOT_Z
SIM_ASPD_RND
SIM_WIND_DIR
SIM_GPS_BYTELOSS
SIM_FLOW_DELAY
```

peterbarker, 2016-12-13:
On Tue, 13 Dec 2016, Ian A Mason wrote:

> Sorry ahead of time for being thick, but that seems to give me a key error:
.
.
> but I still get the key error in the lookup in the if. I looked at the
> dictionary and the
> only SIM_ keys I see are
.
.

You'll need a newer version of ArduPilot:
STABILIZE> param show SIM_SPEEDUP
STABILIZE> SIM_SPEEDUP      -1.000000

This is on
APM: APM:Copter V3.5-dev (5a8a3c5a)

Yours,
-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

hamishwillee, 2016-12-13:
@peterbarker We need to build newer versions of ardupilot!

ianamason, 2016-12-13:
Struth.  Afer a git pull  and rebuilding ardupilot I salivate at:

```
Init APM:Copter V3.5-dev (76106889)
```

But drongokit  won't let me play with it.

```
dronekit-sitl --list
solo-1.2.0
solo-2.0.18
solo-2.0.20
plane-3.3.0
copter-3.3
rover-2.50
```

Which makes me think I understand  this last remark by @hamishwillee.

peterbarker, 2016-12-13:
On Tue, 13 Dec 2016, Ian A Mason wrote:

> Struth. Afer a git pull and rebuilding ardupilot I salivate at:
>
> Init APM:Copter V3.5-dev (76106889)
>
> But drongokit won't let me play with it.
>
> dronekit-sitl --list
> solo-1.2.0
> solo-2.0.18 [...]

Yeah.  Those binaries are a little long-in-the-tooth.  I wonder if we have 
instructions around somewhere for how to update them....

dronekit-sitl can be used to launch a binary you build yourself, here's an 
example which does so:

https://github.com/dronekit/dronekit-python/blob/source-system-filtering/examples/avoidance/avoidance.py#L23

You do need to specify the path to the vehicle defaults file 
(something in Tools/autotest/default_params, typically)

ianamason, 2016-12-13:
You beaut @peterbarker  I'll try that.

hamishwillee, 2016-12-13:
@peterbarker There are instructions, but I have never followed them: https://github.com/dronekit/dronekit-sitl/tree/master/stage#building-a-new-sitl-image

I was hoping someone who had would do the build for us hint hint

ianamason, 2016-12-14:
Wow how old is that PIP package, is it from the future or the past?  In the repo I see:
```
class SITL():
    def __init__(self, path=None, instance=None, defaults_filepath=None, gdb=False, gdb_breakpoints=[], valgrind=False):
```
but in my pip package I see
```
class SITL(object):
    def __init__(self, path=None):
        if path:
            self.path = os.path.realpath(path)
        else:
            self.path = None
        self.p = None
        self.wd = None
```

At least this explains why I have no hair on my head.   Should I be working from the repo
or the pip package?

peterbarker, 2016-12-14:
On Tue, 13 Dec 2016, Ian A Mason wrote:

> class SITL():
>     def __init__(self, path=None, instance=None, defaults_filepath=None, gdb
> =False, gdb_breakpoints=[], valgrind=False):
> 
> but in my pip package I see
> 
> class SITL(object):
>     def __init__(self, path=None):

> At least this explains why I have no hair on my head. Should I be 
> working from the repo or the pip package?

repo.  I've added many toys.

hamishwillee, 2016-12-14:
@peterbarker Can we get a new release of the pip then? The docs suggest various approaches, but I typically use a PIP if one is provided.

ianamason, 2016-12-14:
Bingo. Thanks @peterbarker.

tipehugo, 2017-01-23:
Hi I get the same problem, but I have not understood how you have solved it  :) Can you explain me

ianamason, 2017-01-23:
So if you look at the README.md  of this repository you will see they describe two  APIs.
A simple one, and then a more low level one that involves making a SITL object and manipulating that.  So you build the latest drone from the ardupilot repository and then use the binaries
and default params from that:

```
binary_path = '/home/vagrant/Repositories/ardupilot/build/sitl/bin/arducopter-quad'

params_path = '/home/vagrant/Repositories/ardupilot/Tools/autotest/default_params/copter.parm'

```
 Because I am running all this inside a custom actor system
I had to wrap my sitl instance into a separate process, like so:

```
#!/usr/bin/env python

import sys
import dronekit_sitl

usage = 'Usage:   {0} instanceno binary_path params_path home verbose'

def main(args):
    if len(args) != 6:
        print usage.format(args[0])
        return 1
    else:
        sitl = dronekit_sitl.SITL(instance=int(args[1]), path=args[2], defaults_filepath=args[3])
        largs = ['--home={0}'.format(args[4])]
        verbose = bool(args[5])
        sitl.launch(largs, verbose=verbose)
        sitl.block_until_ready(verbose=verbose)
        code = sitl.complete(verbose=verbose)
        sitl.poll()
        sitl.stop()
        # hpoefully should not see this...
        sys.stderr.write('\n----> SITL exiting <----\n')
        return 0

if __name__ == '__main__':
    sys.exit(main(sys.argv))

```

tipehugo, 2017-01-23:
Okay but I'm using JSBSIM and I would like to do this on a plane. Is it possible ?

ianamason, 2017-01-23:
I think so, building ardupilot requires (I think) building jbsim. To use a plane you would just
build the binary, then use that and the defaults for the plane.

tipehugo, 2017-01-23:
Sorry , I do not understand what means build the binary. You sent me binaries to quadrucopters , not to plane ?

ianamason, 2017-01-23:
you will have to follow the building instructions for ardupilot.
you will build the plane, then locate the binary, which will be in the 

```
ardupilot/build/sitl/bin/
```
directory. Then you will need to find the default params for the plane. They should be 
in 

```
ardupilot/Tools/autotest/default_params/
```

good luck.

ianamason, 2017-01-23:
https://github.com/ArduPilot/ardupilot

tipehugo, 2017-01-23:
Ok so i'm in https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/default_params/plane-jsbsim.parm

I notice that there is not SIM_SPEEDUP param and I have to create it right ? 
thank you for your help

ianamason, 2017-01-23:
No I used the method mentioned above.

```
while True:
    self.vehicle.parameters['SIM_SPEEDUP'] = value
    print("Setting SIM_SPEEDUP")
    if self.vehicle.parameters['SIM_SPEEDUP'] == value:
        break
    time.sleep(0.1)
```

tipehugo, 2017-01-23:
I just need to copy ant to paste a bit of code ? 
And what I have to copy is the code of the param sim_speedup . But have you an idea where I can find it ?

tipehugo, 2017-01-23:
I would like to use   vehicle.parameters['SIM_SPEEDUP']=10 but it does not work actually

ianamason, 2017-01-23:
That is roughly what I do.

tipehugo, 2017-01-23:
Okay I have understood the problematic, I will work on it later. Thank you very much for your help

closed 2017-03-28 by ianamason