dronekit/dronekit-sitl #111: 'dronekit-sitl copter' error

KAVYAP, 2017-05-27:
Hello,

I am trying to run dronekit-sitl copter-3.3, I get this error.

dronekit-sitl copter-3.3 --home=-35.363261,149.165230,584,353
os: linux, apm: copter, release: 3.3
SITL already Downloaded and Extracted.
Ready to boot.
Failed to run (/home/kavyap/.dronekit/sitl/copter-3.3/apm)
Traceback (most recent call last):
  File "/usr/local/bin/dronekit-sitl", line 9, in <module>
    load_entry_point('dronekit-sitl==3.2.0', 'console_scripts', 'dronekit-sitl')()
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 593, in main
    sitl.launch(args, verbose=True)
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 253, in launch
    caps = ArdupilotCapabilities(self.path)
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 165, in __init__
    self.has_defaults_path = "--defaults path" in helptext
UnboundLocalError: local variable 'helptext' referenced before assignment

i am using linux ubuntu 14.04 (32-bit). can anyone help me

imutkarshpatil, 2017-10-24:
I'm facing the same issue while running dronekit-sitl with copter-3.3 on Debian Linux.

imutkarshpatil, 2017-10-24:
@KAVYAP are you using dronekit-sitl on Raspberry Pi or any otherARM-based system? If yes then I guess this won't work, found this mentioned here in #95