dronekit/dronekit-sitl #113: Dronekit-sitl download error: platform cygwin is not supported

ccnavio, 2017-06-21:
Collecting git+https://github.com/dronekit/dronekit-sitl
  Cloning https://github.com/dronekit/dronekit-sitl to /tmp/pip-ZUGPvF-build
Collecting psutil>=3.0 (from dronekit-sitl==3.2.0)
  Using cached psutil-5.2.2.tar.gz
    Complete output from command python setup.py egg_info:
    platform cygwin is not supported

Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-EwWyZ3/psutil/

How do I go around this so I can download Dronekit-SITL?

hamishwillee, 2017-06-21:
Don't use cygwin - SITL allows you to download native binaries and runs on the Windows command line.

ji5u1031, 2018-04-24:
hamishwillee
can you tell me how can i run SITL on the Windows command line without cygwin??
i got a lot of trouble with this...
once I run SITL on the command line, it says 'sim_vehicle.py' is not available.
It would be pleasure if you tell me how to resolve this or how to install SITL without cygwin.
It was only way introduced on Ardupilot Webpage

hamishwillee, 2018-04-25:
@ji5u1031 I would actually install DK-SITL from Github master (the second option shown here: https://github.com/dronekit/dronekit-sitl#installing ). DK-SITL does not need cygwin.

Instructions for using SITL can be found here: http://python.dronekit.io/guide/quick_start.html  (don't install from PIP).