dronekit/dronekit-sitl #114: Changing SYSID_THISMAV

thefrankinator7, 2017-07-06:
I'm trying to connect two instances of SITL to QGroundControl with MAVProxy, but when I connect to both at the same time, the connection keeps switching back and forth between the two. I believe this is because both instances have the parameter SYSID_THISMAV set to 1. However, I cannot change the parameter in MAVProxy with either the 'param set' or 'param load' command. I tried changing the default.parm file in dronekit/sitl/copter-3.3 but that didn't work either. Is this the wrong file? Is there another way to change the ID for one of the instances of SITL?
Any help would be appreciated.
Thanks in advance.

peterbarker, 2017-07-17:
On Thu, 6 Jul 2017, thefrankinator7 wrote:

> I'm trying to connect two instances of SITL to QGroundControl with MAVProxy,
> but when I connect to both at the same time, the connection keeps switching
> back and forth between the two. I believe [...]

A reboot is required for that parameter change to take effect.

You may wish to look at the "avoidance" example here:

https://github.com/peterbarker/dronekit-python/tree/source-system-filtering/examples/avoidance

Peter

alduxvm, 2018-04-26:
Hi Peter, is there an implemented way without using your example? thanks!

peterbarker, 2018-04-26:
On Wed, 25 Apr 2018, Aldo Vargas wrote:

> Hi Peter, is there an implemented way without using your example? thanks!

Not as far as I know.

alduxvm, 2018-04-26:
thanks Peter!

zarar7576, 2019-08-19:
You can solve the issue by using pre-compiled binaries with different SYSIDs I will soon make a repository about this because a lot of people seem to have this issue. Have a look here I have uploaded some binaries that may help you out  #126