dronekit/dronekit-sitl #115: dronekit-sitl: command not found

welkerCode, 2017-08-09:
I am managed to get dronekit and dronekit-sitl with dronekit-python working on a virtual machine running Ubuntu, but trying to do the same thing on a Kubuntu system, I get the following error when trying to run dronekit-sitl copter:

dronekit-sitl: command not found.

I followed the instructions with using pip to install dronekit and dronekit-sitl.  Any idea what could be causing this?

hamishwillee, 2017-08-09:
That means what it says - dronekit-sitl is not installed. Can you see it when you do pip list (or pip3 list)? This sort of thing can happen in Python when you have multiple versions and you install to a different python than you think you have.

Things I would try include reinstalling, installing from source, running a script that types our which python env you are running (and making sure it is the same as you think you are.). Upshot, installation problem not a DK problem as such.

welkerCode, 2017-08-10:
When I use pip list, both dronekit and dronekit-sitl appear, and I only have one version of Python on this machine (2.7.12).  I have tried re-installing through pip several times, but I haven't tried installing from source yet.  I will try that tomorrow.  Thanks for the tip!

hamishwillee, 2017-08-10:
Every time I've had a problem like this it was 2 versions of Python. The way I have shown this is by doing `python -v` in terminal and running a script `python myscript.py` (that prints the python runtime). If they give different results then you do have multiple versions. Anyway, good luck - very frustrating to get stuck at the first steps.

welkerCode, 2017-08-10:
You were right.  I had Python3 installed.  After uninstalling both, and reinstalling python 2.7.12 and pip using the sudo easy_install pip (as recommended on the dronekit wiki), uninstalling and reinstalling dronekit and dronekit-sitl again, dronekit appears to be working now.  Thanks for your help hamishwillie!

hamishwillee, 2017-08-10:
> You were right

I love those words :-) You're most welcome.

closed 2017-08-10 by hamishwillee