dronekit/dronekit-sitl #118: Command 'dronekit-sitl copter' hangs on 'waiting for connection'

Martin-Pozniak, 2017-10-04:
So I understand that when you run the command "dronekit-sitl copter" The first thing it does is try to establish a connection to 127.0.0.1:5760. For some reason, the script can't connect to that TCP IP and Port.
[image]

Shown in the photo above is the output which hangs on "waiting for connection"

I have tried scanning for the port and it looks like the port doesn't even exsist.(shown below is an image of an nmap scan of the closed ports on 127.0.0.1. Right where you think 5760 would be it skips from 5730 to 5800.

[image]

The nmap scan also reveals that there are only two ports open on 127.0.0.1

[image]

I'm running dronekit-sitl on windows 10. I have already added new inbound and outbound rules on the firewall for the TCP port 5760.

I also tried the command on different networks to no avail.

I have run out of ideas to solve the issue and am in need of help!

lucetewoo, 2017-11-16:
did you solve the that problem? i also have same problem... 
if you solved that problem, tell me how to solve that problem....i need your help...!!!

Martin-Pozniak, 2017-11-17:
@lucetewoo Hello, I did manage to solve this problem.

Turns out we are using the sim wrong. The reason it is stuck on waiting for connection is that the simulated vehicle from dronekit-sitl is waiting for some code to connect to it and use it! If nothing connects to it to use it, of course, it will just sit there and wait for a connection. Observe the following steps to understand whats going on.

Step 1)
[image]
Notice 2 terminals open (one running the dronekit-sitl simulated copter, one prepared to run python code) and one python script using drone kit API to simulate a basic takeoff. In the circled terminal we have initialized the sitl copter and it is waiting to be connected to by a vehicle object.

Step 2)
[image]
Notice in the second terminal we are about to run the basic_takeoff python script. Once we run this, the code will connect to the sitl copter in the first terminal.

Step 3)
[image]
This line of code is what actually connects the vehicle object to the sitl copter which is currently "waiting for connection". In real life, you would change this connect function to connect to some real physical drone

Step4)
[image]
Observe Now, we ran the script in the second terminal (1) and as soon as we did that the simulated vehicle bound its port and connected to our code (2). The script ran and the simulator worked!

_P.S. At the top of that python script (which you can find on google) there needs to be an 'import time" statement otherwise the script won't recognize time.sleep()_ I added this before step 4 was executed.

Hopefully this helps you

Hussainali819, 2021-03-14:
till step 3, everything worked well but after running the same code as provided in step 3, I get following error
CRITICAL:autopilot:APM:Copter V3.3 (d6053245)
CRITICAL:autopilot:Frame: QUAD
CRITICAL:autopilot:Calibrating barometer
CRITICAL:autopilot:Initialising APM...
CRITICAL:autopilot:barometer calibration complete
CRITICAL:autopilot:GROUND START
Basic pre-arm checks
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
Arming motors
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...
ERROR:autopilot:ARMING MOTORS
CRITICAL:autopilot:GROUND START
 Waiting for arming...
CRITICAL:autopilot:Initialising APM...
 Waiting for arming...
Taking off!
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
ERROR:autopilot:DISARMING MOTORS
 Altitude:  0.02
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
and then it continues, and no changes occur in mission planner

I'm using windows 10
python 3.9.2
mission planner 1.3.71.1
dronekit 2.9.2
dronekit-sitl 3.3.0
pymavlink 2.4.14
MAVProxy 1.8.34

kindly help...