dronekit/dronekit-sitl #122: dronekit-sitl copter: command not found

duckhang9113, 2018-02-03:
I followed the instruction and install mavproxy, dronekit and, dronekit-sitl but when I try to run dronekit-sitl copter the return is command not found. I checked pip list both are installed. I ran python --version the version I'm using is 2.7.14. How can I fix this?

SarveshTJ, 2018-02-05:
Hey, I am having the same problem.

Do we have to change directories or something like that?

SarveshTJ, 2018-02-05:
In the dronekit-python Documentation, it's mentioned that DroneKit-SITL runs for x86 version of Linux only. I am using x64 version. Is this the issue?

http://python.dronekit.io/develop/sitl_setup.html

AnthonySong98, 2018-04-20:
Same problem ！！！Any thoughts yet？

zengdiqing1994, 2018-07-18:
but where to find x86 version of linux?

hamishwillee, 2018-07-19:
The toolchain works by downloading prebuilt images. They are downloaded when you call a command like `dronekit-sitl copter`.

BUT, the images are only prebuilt for x86 and for quite old versions of ArduPilot. If you want a more recent version of SITL OR if you want to have x64 then you can still use DK-SITL, but you will have to build the binaries yourself following the normal instructions for building native binaries.

If you do this you lose most of the benefit of DK-SITL - getting easy pre-build binaries. However this is still useful for integrating with test frameworks etc.

ABDULBASIT1998, 2019-03-16:
> The toolchain works by downloading prebuilt images. They are downloaded when you call a command like `dronekit-sitl copter`.
>
> BUT, the images are only prebuilt for x86 and for quite old versions of [...]

how to run the simulator?any idea i am x64 same problem? where to find the simulator file?

hamishwillee, 2019-03-18:
@ABDULBASIT1998 You just quoted the answer to your own question. The files are not provided. 

What you can do is build SITL natively: http://ardupilot.org/dev/docs/building-the-code.html

Even though you can't use prebuilt binaries, some people will still want to use DK-SITL with the binaries they build natively, because that makes it easier to script tests etc. You can do that by passing the file path to your natively built binary.

hamzawinix, 2019-08-24:
this solved it for me
` sudo -H pip install dronekit-sitl
`