dronekit/dronekit-sitl #124: Timeout on <Calibrating barometer>

DylanVanAssche, 2018-03-02:
I'm trying to get the quick start example (http://python.dronekit.io/guide/quick_start.html) working but every time I want to start an SITL copter it fails due a timeout when calibrating the barometer.

**Code:**
```
print ("Start simulator (SITL)")
import dronekit_sitl
sitl = dronekit_sitl.start_default()
connection_string = sitl.connection_string()

# Import DroneKit-Python
from dronekit import connect, VehicleMode

# Connect to the Vehicle.
#print("Connecting to vehicle on: %s" % (connection_string,))
print("Connecting to vehicle on: {0}".format(connection_string))
vehicle = connect(connection_string, wait_ready=True,  heartbeat_timeout=180)

# Get some vehicle attributes (state)
print("GPS {0}".format(vehicle.gps_0))
print("Battery {0}".format(vehicle.battery))
print("Heartbeat {0}".format(vehicle.last_heartbeat))
print("Armable {0}".format(vehicle.is_armable))
print("System state {0}".format(vehicle.system_status.state))
print("Mode {0}".format(vehicle.mode.name ))

# Close vehicle object before exiting script
vehicle.close()

# Shut down simulator
sitl.stop()
print("Completed")
```

**Output:**
```
Start simulator (SITL)
Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
Connecting to vehicle on: tcp:127.0.0.1:5760
>>> APM:Copter V3.3 (d6053245)
>>> Frame: QUAD
>>> Calibrating barometer
Traceback (most recent call last):
  File "drone.py", line 12, in <module>
    vehicle = connect(connection_string, wait_ready=True,  heartbeat_timeout=180)
  File "/usr/lib/python3.4/site-packages/dronekit/__init__.py", line 2849, in connect
    vehicle.wait_ready(True)
  File "/usr/lib/python3.4/site-packages/dronekit/__init__.py", line 2199, in wait_ready
    timeout)
dronekit.APIException: wait_ready experienced a timeout after 30 seconds.
```

**Environment:**

- OpenSUSE Leap 42.3 / Xubuntu 14.04 LTS
- Python 3.4
- Intel CPU x64 / AMD CPU x64

Am I doing something wrong? I read that the SITL images are only for x86 architecture, but I run the x64 architecture...

DylanVanAssche, 2018-03-04:
Hmmm Python 3 isn't supported, I missed that in the Quick Start guide.

closed 2018-03-04 by DylanVanAssche

sarthak268, 2018-05-01:
Please add this to the README also, it would be helpful.

Vaibhavg133, 2019-10-03:
I'm also facing the same issue in python2.7 please hel
[image]
p