dronekit/dronekit-sitl #126: Multiple SITLs in a Computer

mustaphos, 2018-03-10:
Hi.
I want to run multiple SITL copters in one computer.
How can i change the 5760 TCP port when initializing the SITL?
For example:
Drone1 : 5760
Drone2: 5761 and so on...
Regards...

Javiercerna, 2018-03-18:
Hi,

You could use the "I" option to use 5770, 5780 and so on... It works in multiples of 10, so I don't think you would be able to use 5761 in this way.

`dronekit-sitl copter -I1` # Starts a model in 5770
`dronekit-sitl copter -I2` # Starts a model in 5780

From the dronekit-sitl command help:

`--instance N       set instance of SITL (adds 10*instance to all port numbers)`

alduxvm, 2018-04-26:
yes, the instance number works for dronekit-silt, and then you can add a mavproxy with the relevant ports and even change the sysmavid in each simulated vehicle, but they are not unique vehicles, so the ground station "jumps" from one device to another in a fast way... is there a way to avoid this without using the peterbaker multisitl solution?

FloodChen, 2018-05-30:
@alduxvm i have  the same question with you, do you find any solution?

alduxvm, 2018-05-30:
Hi, I haven't found a solution... changing the ID of mavlink requieres reboot, I had one idea of compiling an .hex with different ID and then use that one with dronekit silt, but I did not reached to far. sorry

mustaphos, 2018-05-30:
Well the worst but working solution for that is multiple virtual machines in one host computer. 
Then start sitl in each virtual machine.

peterbarker, 2018-06-01:
On Wed, 30 May 2018, 洪臣 wrote:

> @alduxvm i have the same question with you, do you find any solution?

https://github.com/dronekit/dronekit-python/tree/source-system-filtering/examples/avoidance

vasantharajr, 2019-03-19:
https://github.com/vasantharajr/Autonomous_UAVs_Swarm_Mission

vasantharajr, 2019-03-19:
system id can be changed in mavproxy 
mavproxy.py --master=0.0.0.0:14550 --master=0.0.0.0:14560 --source-system=1 --console --map

zarar7576, 2019-08-19:
I solved this issue by compiling binaries with SYSID set to different numbers. The zip file below contains binaries with SYSID from 1-3 it also contains the default eeprom and pram files. You need to  run the below commands in the same folder to run multiple instances of SITL with different SYSIDs.

[firmwares.zip](https://github.com/dronekit/dronekit-sitl/files/3514665/firmwares.zip)

```
dronekit-sitl ./sysid_1 --home=-35.363262, 149.165230,584,0 &

dronekit-sitl ./sysid_2 -I1 --home=-35.363352, 149.165123,584,0 &

dronekit-sitl ./sysid_3 -I2 --home=-35.363348, 149.165346,584,0 &
```

(Remove "&" at the end for non-Linux users and run each command in a new cmd prompt(windows) I don't know the mac way sorry  you can also create a bash file and run them together)

The above commands will generate three different quads at 5760, 5770 and 5780 you can directly connect them to your ground control station(I used Q ground control) or use MAVproxy for multiple outputs like below

```
mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551 

mavproxy.py --master tcp:127.0.0.1:5770 --sitl 127.0.0.1:5502 --out 127.0.0.1:14552 --out 127.0.0.1:14553 

mavproxy.py --master tcp:127.0.0.1:5780 --sitl 127.0.0.1:5503 --out 127.0.0.1:14554 --out 127.0.0.1:14555
```
(dont make a bash file for these just run them in seprate terminals. BTW if you are able to make a bash that can run all these at the same time plss do let me know)

aviontics, 2020-03-12:
This seems to result in 'no link' issue.
It also produces 'PANIC: AP_Baro::read unsuccessful for more than 500ms in AP_Baro::calibrate [2]
' when mavproxy was run.

command
mavproxy.py --master tcp:127.0.0.1:5770 --sitl 127.0.0.1:5502 --out 192.168.0.5:14552 --out 127.0.0.1:14553 --cmd="param load /home//dronekit-sitl/default.parm"

floridatechnologies, 2020-05-04:
> This seems to result in 'no link' issue.
> It also produces 'PANIC: AP_Baro::read unsuccessful for more than 500ms in AP_Baro::calibrate [2]
> ' when mavproxy was run.
>
> command
> mavproxy.py --master [...]

Any updates on this

avcuenes, 2020-09-12:
> I solved this issue by compiling binaries with SYSID set to different numbers. The zip file below contains binaries with SYSID from 1-3 it also contains the default eeprom and pram files. You need to run the below commands in the same folder to run multiple instances of SITL with different SYSIDs.
> 
> [firmwares.zip](https://github.com/dronekit/dronekit-sitl/files/3514665/firmwares.zip)
> 
> ```
> dronekit-sitl ./sysid_1 --model=quad --home=-35.363262,149.165230,584,0 &
> 
> dronekit-sitl ./sysid_2  --model=quad  -I1 --home=-35.363352,149.165123,584,0 &
> 
> dronekit-sitl ./sysid_3  --model=quad  -I2 --home=-35.363348,149.165346,584,0 &
> ```
> 
> (Remove "&" at the end for non-Linux users and run each command in a new cmd prompt(windows) I don't know the mac way sorry you can also create a bash file and run them together)
> 
> The above commands will generate three different quads at 5760, 5770 and 5780 you can directly connect them to your ground control station(I used Q ground control) or use MAVproxy for multiple outputs like below
> 
> ```
> mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551 
> 
> mavproxy.py --master tcp:127.0.0.1:5770 --sitl 127.0.0.1:5502 --out 127.0.0.1:14552 --out 127.0.0.1:14553 
> 
> mavproxy.py --master tcp:127.0.0.1:5780 --sitl 127.0.0.1:5503 --out 127.0.0.1:14554 --out 127.0.0.1:14555
> ```
> 
> (don't make a bash file for these just run them in separate terminals. BTW if you are able to make a bash that can run all these at the same time please do let me know)
> 
> Just FYI: you will need to load the param from default.parm file in MAVproxy
> 
> `param load default.parm`
> 
> default.parm comes in the firmware folder also I disabled the battery while compiling the binaries because I was too annoyed by the low battery of my simulated drone.
> `

you can use like this. copy this to text file and save as .bat . after that you can use easy 

start cmd /k dronekit-sitl copter --home=39.945102,32.593052,0,180 
start cmd /k mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551 
start cmd /k dronekit-sitl copter -I1 --home=39.945102,32.593052,0,180
start cmd /k mavproxy.py --master tcp:127.0.0.1:5770 --sitl 127.0.0.1:5502 --out 127.0.0.1:14552 --out 127.0.0.1:14553
start cmd /k dronekit-sitl copter -I2 --home=39.945102,32.593052,0,180 
start cmd /k mavproxy.py --master tcp:127.0.0.1:5780 --sitl 127.0.0.1:5503 --out 127.0.0.1:14554 --out 127.0.0.1:14555

arthurrichards77, 2020-10-19:
Apologies if this is old news: but having fought with this same issue for a while, I was delighted to find a "--sysid" option in SITL.  It is documented but easy to miss in the usage list.  So, the following (in four different terminals) gives me a multiplexed MAVLINK stream on UDP port 14553, and I can then use QGroundControl in multiple UAV mode but with just a single UDP connection.  Each SITL has an accompanying mavproxy to forward its MAVLINK to the UDP port.

```
arducopter --model=quad
mavproxy.py --master=tcp:127.0.0.1:5760 --out=udp:127.0.0.1:14553
arducopter --model=quad --instance=1 --sysid=2
mavproxy.py --master=tcp:127.0.0.1:5770 --out=udp:127.0.0.1:14553
```