dronekit/dronekit-sitl #138: Running dronekit-sitl with Webots in Windows

PrasadNR, 2019-10-25:
In reference to https://github.com/ArduPilot/ardupilot/pull/12006#issuecomment-544224951 and https://github.com/ArduPilot/ardupilot/pull/12006#issuecomment-546190368 , I was looking into the integration of dronekit and Webots. In Webots, I have a [DJI Mavic 2 Pro](https://www.cyberbotics.com/doc/guide/mavic-2-pro?version=develop) with sensors and four motor velocities for actuation.

This is what I am experimenting with:
```

from controller import *
import mavic2proHelper
import cv2

TIME_STEP = 64
TAKEOFF_THRESHOLD_VELOCITY = 163

robot = Robot()

[frontLeftMotor, frontRightMotor, backLeftMotor, backRightMotor] = mavic2proHelper.getMotorAll(robot)
MAX_PROPELLER_VELOCITY = int(frontLeftMotor.getMaxVelocity())

timestep = int(robot.getBasicTimeStep())
mavic2proMotors = mavic2proHelper.getMotorAll(robot)
mavic2proHelper.initialiseMotors(robot, 0)
mavic2proHelper.motorsSpeed(robot, TAKEOFF_THRESHOLD_VELOCITY, TAKEOFF_THRESHOLD_VELOCITY, TAKEOFF_THRESHOLD_VELOCITY, TAKEOFF_THRESHOLD_VELOCITY)

gps = GPS("gps")
gps.enable(TIME_STEP)

gyro = Gyro("gyro")
gyro.enable(TIME_STEP)

while (robot.step(timestep) != -1):

	gpsLocation = gps.getValues()
	gyroValues = gyro.getValues()
	print(gyroValues)

```
```
def getMotorAll(robot):
	frontLeftMotor = robot.getMotor('front left propeller')
	frontRightMotor = robot.getMotor('front right propeller')
	backLeftMotor = robot.getMotor('rear left propeller')
	backRightMotor = robot.getMotor('rear right propeller')
	return [frontLeftMotor, frontRightMotor, backLeftMotor, backRightMotor]

def motorsSpeed(robot, v1, v2, v3, v4):
	[frontLeftMotor, frontRightMotor, backLeftMotor, backRightMotor] = getMotorAll(robot)
	frontLeftMotor.setVelocity(v1)
	frontRightMotor.setVelocity(v2)
	backLeftMotor.setVelocity(v3)
	backRightMotor.setVelocity(v4)
	return

def initialiseMotors(robot, MAX_PROPELLER_VELOCITY):
	[frontLeftMotor, frontRightMotor, backLeftMotor, backRightMotor] = getMotorAll(robot)
	frontLeftMotor.setPosition(float('inf'))
	frontRightMotor.setPosition(float('inf'))
	backLeftMotor.setPosition(float('inf'))
	backRightMotor.setPosition(float('inf'))

	motorsSpeed(robot, 0, 0, 0, 0)
	return
```

But, what I am not sure is how I can make the drone takeoff. Beyond 163, drone starts flipping. I am not sure how this drone can be connected to the mavlink interface that dronekit-sitl supports.

HefnySco, 2019-10-25:
This is why I had to make a less good looking quadcopter.
Although this quadcopter has great looking the motor torque and velocity is not made to be compatible with values sent from Ardupilot signals.
You need to do the math to recalculate torque and thrust so that servos values - or multipliers- suitable for flying it. this is the main issue I faced when I was dealing with this quad the time I developed the integration.
I made this [excel](https://docs.google.com/spreadsheets/d/1ZXIDLAhBHd2HfScILtGQehjGkqyf3NcTf5lo6G59MAk/edit?usp=sharing) to help me to calculate this.

PrasadNR, 2019-10-25:
@HefnySco Ok. I guess you have confirmed my assumption regarding motor velocities. I am closing this issue. However, I am not able to get anything running properly in Webots in Windows. Could you guide me as to how your simulator could be run? (Maybe by running your simulations inside something like this https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/ ?)

P.S.: If anyone else believes that simulation with DJI Mavic 2 Pro could be done, please feel free to reopen this issue. I believe this is not possible and hence I am closing this issue.

closed 2019-10-25 by PrasadNR