dronekit/dronekit-sitl #142: Cant change flight mode on for drone kit sitl

Pilotben, 2020-05-15:
Hi, I am new to DroneKit and I just downloaded DroneKit and MAVProxy to my Mac but I have an issue where I initialise a SITL drone and then open mavproxy and connect to it and I can see a connection is established however none of the commands I type in the MAVProxy terminal are executed please see attached picture. I would really appreciate any help on this topic
[image]

itzmejawad, 2020-09-11:
I am experiencing the same problem like yours in Ubuntu... even i downgraded and checked with different OS distributions but nothing works

thianesh, 2020-09-14:
I too

itzmejawad, 2020-09-14:
Hi @Pilotben 

As mentioned by @zarar7576 in [https://github.com/dronekit/dronekit-sitl/issues/105](url)

I have downloaded [https://github.com/dronekit/dronekit-sitl/files/3514761/firmware_36.zip](url)

once downloaded copy the contents to ~/.dronekit/sitl/copter-3.3 directory and rename the **"copter_36"** file to **"apm"**

and also change the permission of that folder by doing 
**sudo chmod -R a+rwx ~/.dronekit/sitl/copter-3.3/**

it will work

the mode change problem will be solved

Mettikodeva-Nara, 2021-11-03:
> Hi @Pilotben @thianesh
> 
> As mentioned by @zarar7576 in #105
> 
> I have downloaded https://github.com/dronekit/dronekit-sitl/files/3514761/firmware_36.zip
> 
> once downloaded copy the contents to ~/.dronekit/sitl/copter-3.3 directory and rename the **"copter_36"** file to **"apm"**
> 
> and also change the permission of that folder by doing **sudo chmod -R a+rwx ~/.dronekit/sitl/copter-3.3/**
> 
> it will work
> 
> the mode change problem will be solved

how to do this in windows?
thanks